dresden-elektronik/deconz-rest-plugin #7921: Some AirQuality sensor values (_TZE200_dwcarsat) are not passed on to IOBroker after updating to 2.28.1

theflyer08, 2024-09-10:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I own an air quality sensor _TZE200_dwcarsat (DDF #7146). The air quality sensor worked perfectly for months. With the latest update 2.28.1, I had to reintegrate the sensor into Phoscon because no connection to the Zigbee hub (Conbee II) could be established. After the reintegration, all sensor values were back in Phoscon and are updated regularly in Phoscon. However, only the following sensor values are now passed on to IOBroker: CO2, Formaldehyde & Particulate Matter. TVOC, Temperature & Humidity are missing as objects in IOBroker.

### Steps to reproduce the behavior

The behaviour is consistent with 2.28.1. All sensor values can be seen in Phoscon and are regularly updated there. Some of the corresponding objects in IOBroker are missing (TVOC, Temperature & Humidity).

### Expected behavior

All  sensor values are passed on to IOBroker and are regularly updated.

### Screenshots

[image]

### Environment

- Host system: Debian 11
- Running method: Proxmox LXC
- Firmware version: 26780700
- deCONZ version (not Home assistant Addon version!): 2.28.1
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? yes, SSD

Mimiix, 2024-09-10:
Hi,

Can you show the api output under help, api information?

theflyer08, 2024-09-10:
@Mimiix: By checking the api information I figured out that IOBroker is actually not renaming the object correctly. So my AirQuality sensor is named  IKEA Tadfri Switch. However, all sensor values are present. Thank you for guiding me in the right direction. 

I will close the issue.

closed 2024-09-10 by theflyer08