dresden-elektronik/deconz-rest-plugin #7925: Unable to add 4 button sunrichter/NAMRON AS switch 4512719

olemr, 2024-09-14:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

As described in https://forum.phoscon.de/t/namron-4512719-no-longer-able-to-add/5066
Was directed here.
Got logs now.

### Steps to reproduce the behavior

1. Factory resetting the 4512719 switch
2. Start an Add new switch search in Phoscon
3. Start add procedure on the 4512719
4. The Add fails.

### Expected behavior

I expected the switch to be added as it did in the past.
Not sure when it was broken.

### Screenshots

Screenshots in forum:
https://forum.phoscon.de/t/namron-4512719-no-longer-able-to-add/5066

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian Debian install
- Firmware version: (26400500)
- deCONZ version: 2.28.1
- Device: ConBee I
- Do you use an USB extension cable: yes, with a powered HUB
- Are there any other USB or serial devices connected to the host system? No

### deCONZ Logs

`
15:09:01:134 MAC Poll 0x02 0x6470, life time: 4294967295 sec, dev timeout: 4294967295 sec
15:09:01:470 Beacon src: 0xBB69 ch: 25 updateId: 1
15:09:01:486 Beacon src: 0xB706 ch: 25 updateId: 1
15:09:01:486 Beacon src: 0xFF8D ch: 25 updateId: 1
15:09:01:487 Beacon src: 0x87FA ch: 25 updateId: 1
15:09:01:487 Beacon src: 0x7E37 ch: 25 updateId: 1
15:09:01:487 Beacon src: 0x1788 ch: 25 updateId: 1
15:09:01:487 Beacon src: 0xC780 ch: 25 updateId: 1
15:09:01:502 Beacon src: 0x2EE5 ch: 25 updateId: 1
15:09:01:502 Beacon src: 0x7968 ch: 25 updateId: 1
15:09:01:503 Beacon src: 0xF91F ch: 25 updateId: 1
15:09:01:503 Beacon src: 0x1B14 ch: 25 updateId: 1
15:09:01:503 Beacon src: 0x67F5 ch: 25 updateId: 1
15:09:01:503 Beacon src: 0x782E ch: 25 updateId: 1
15:09:01:518 Beacon src: 0x2989 ch: 25 updateId: 1
15:09:01:518 Beacon src: 0xD079 ch: 25 updateId: 1
15:09:01:518 Beacon src: 0xC387 ch: 25 updateId: 1
15:09:01:519 Beacon src: 0x3395 ch: 25 updateId: 1
15:09:01:534 Beacon src: 0x67A3 ch: 25 updateId: 1
15:09:01:534 Beacon src: 0x1B8D ch: 25 updateId: 1
15:09:01:535 Beacon src: 0xAEE0 ch: 25 updateId: 1
15:09:01:550 Beacon src: 0xF440 ch: 25 updateId: 1
15:09:01:550 Beacon src: 0xF867 ch: 25 updateId: 1
15:09:01:566 Beacon src: 0xFEE7 ch: 25 updateId: 1
15:09:01:566 Beacon src: 0xDB48 ch: 25 updateId: 1
15:09:01:566 Beacon src: 0xB196 ch: 25 updateId: 1
[... 484 lines cut ...]
      "src.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "groupcast",
      "config.group": 1,
      "src.ep": 2,
      "cl": "0x0006"
    },
    {
      "bind": "groupcast",
      "config.group": 1,
      "src.ep": 2,
      "cl": "0x0008"
    }
  ]
}`

Tried adding NAMRON  AS with 2 spaces as hinted to in forum. No joy.

SwoopX, 2024-09-15:
Hi,

thanks for raising your matter here, but it looks as I stand right with my initial assumption on the manufacturer name, so this is not a bug. Therefore, I'll close this issue.

[image]

closed 2024-09-15 by SwoopX

olemr, 2024-09-15:
@SwoopX what should that PR contain?
Do I have to clone, build and test a fix myself and then include that fix in a PR?
Never done this before ...

olemr, 2024-09-15:
Also please note, that this is not a new switch. It used to work fine a few years back, probably pre DDF ..

SwoopX, 2024-09-15:
You just need to add modelid and manufacturer to a fitting DDF like here: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7889/files

That's it already

olemr, 2024-09-15:
In the original sunrichter/4button_switch.json that looks like this:
  "manufacturername": [
    "EcoDim",
    "EcoDim",
    "Sunricher",
    "NAMRON AS",
    "NAMRON AS",
    "ROBB smarrt",
    "ROBB smarrt"
  ], 

I tried adding 
  "manufacturername": [
    "EcoDim",
    "EcoDim",
    "Sunricher",
    "NAMRON AS",
    "NAMRON AS",
    "NAMRON  AS",
    "NAMRON  AS",
    "ROBB smarrt",
    "ROBB smarrt"
  ],
But that did not help.
I'm confused by the double entries there.

Perhaps try:
  "manufacturername": [
    "EcoDim",
    "EcoDim",
    "Sunricher",
    "NAMRON AS",
    "NAMRON  AS",
    "ROBB smarrt",
    "ROBB smarrt"
  ],
 
?

olemr, 2024-09-15:
@SwoopX 
Sorry for mentioning.
I don't think the DDF is picked up at all: (tried my Perhaps try in my latest post, but still no joy)
3489: 15:09:08:464 DEV modelId: 4512719, 0x842E14FFFE90135F
3490: 15:09:08:464 DEV no DDF for 0x842E14FFFE90135F, modelId: 4512719
3491: 15:09:08:465 DEV create on-the-fly DDF for 0x842E14FFFE90135F
3492: 15:09:08:465 DEV enter passive state 0x842E14FFFE90135F

How can we check the content of this on-the-fly DDF? 
Or better still, how to get it to pick up the DDF?

olemr, 2024-09-15:
[image]

olemr, 2024-09-15:
Think I just understood the mapping between manufacturername and modelid, but it looks ok?
```
  "manufacturername": [
    "EcoDim",
    "EcoDim",
    "Sunricher",
    "NAMRON AS",
    "NAMRON  AS",
    "ROBB smarrt",
    "ROBB smarrt"
  ],
  "modelid": [
    "ED-10012",
    "ED-10013",
    "ZG2833K4_EU06",
    "4512719",
    "4512729",
    "ROB_200-008",
    "ROB_200-008-0"
  ],
```

SwoopX, 2024-09-15:
I'm sorry, but you're not thoroughly looking at the provided example and what I've written. You must add manufacturer and modelID, you were (at least initially) adding just the manufacturer. Both values are a json array and the index values at a certain position represent a touple, a valid model/manufacturer combination.

You later on removed the working combination for the 4512719 and replaced the manufacturer at that position with yours. That would mean, everybody else having that combination cannot pair the device anymore. Therefroe, please add your combination at the end.

The legacy code which made the pairing work back in the days has already been removed in favor of leveraging more defined DDFs.