dresden-elektronik/deconz-rest-plugin #7926: Tuya Neo ZigBee Smart Sirene Alarm 90dB Sound Licht Home Security Alarm

extreme4u, 2024-09-14:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

NAS-AB02B2

### Manufacturer

Neo

### Model identifier

TS0601

### Device type to add

Siren

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2024-09-14:
Can you please add the correct manufacturer name and model id in the template?

extreme4u, 2024-09-14:
How to add the name and model? I just can read the config but the write button is not available. 

[image]

Mimiix, 2024-09-15:
> How to add the name and model? I just can read the config but the write button is not available.
> 
> https://zigbee.blakadder.com/Neo_NAS-AB02B2.html
> 
> <img alt="Bildschirmfoto 2024-09-14 um 23 14 51" width="592" src="https://private-user-images.githubusercontent.com/59170539/367544750-6d9e2db8-e561-4a19-9ce5-5ab52f48e708.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYzOTEzNTQsIm5iZiI6MTcyNjM5MTA1NCwicGF0aCI6Ii81OTE3MDUzOS8zNjc1NDQ3NTAtNmQ5ZTJkYjgtZTU2MS00YTE5LTljZTUtNWFiNTJmNDhlNzA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE1VDA5MDQxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkODVkZTZhYTY5NzRhOGU0MGRhNDk2NDQ2NTAwNmE0YmRkZjZkNGRhMzJkNDY1MTUzMDlmNzcyMWUzOTViNDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.mPD_MdM7HZmkKYEnPIzBW4Va5ZMMyUl4uTY6KpEPL7s">

Just put the model id and manufacturer name from the basic cluster in your first post instead of neo.

extreme4u, 2024-09-15:
Okay i clould but I found another post here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6112

Maybe it's already integrated? I try to import this DDF will write if I know more.

closed 2024-10-15 by github-actions[bot]