dresden-elektronik/deconz-rest-plugin #7927: DDFs not loaded on deCONZ restart through API

ebaauw, 2024-09-15:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Sometimes, when restarting deCONZ through the API, deCONZ won't load any DDFs.  Consequently, devices no longer exposed by legacy code are missing from the API, and devices still exposed by legacy code miss the advanced features from the DDF.

After exiting deCONZ (from the GUI) and restarting it, the DDFs are loaded again.

### Steps to reproduce the behavior

Issue a POST to `/config/restartapp` and watch the GUI after deCONZ has restarted.

I've seen the issue a couple of times yesterday, but this morning, trying to get a screenshot, it doesn't appear, of course.  I suspect there might be some sort of racing condition on startup, see also #6710.

### Expected behavior

DDFs should be applied.

### Environment

- Host system: Raspberry
- Running method: Raspbian
- Firmware version: 26780700
- deCONZ version (not Home assistant Addon version!): 2.28.1
- Device: RaspBee II

SwoopX, 2024-09-15:
Hm, interesting case. If I'm not mistaken, manup just recently added some additional debug output where any DDF loading could go south. Maybe you can share the first couple of seconds of deconz' startup log hoping that gives us more valuable insights.

ebaauw, 2024-09-15:
What debug options do you need for that?

SwoopX, 2024-09-15:
> What debug options do you need for that?

I'd say the usual suffices: `--dbg-error=2 --dbg-info=2 --dbg-ddf=1 --dbg-dev=1`

ebaauw, 2024-09-15:
Attached log except when restarting deCONZ through API (after which DDFs wouldn't be loaded) vs restarting deCONZ manually afterwards (after which DDFs are loaded).
[log.gz](https://github.com/user-attachments/files/17005434/log.gz)

Only running with `--http-port=80 --dbg-ddf=1 --dbg-error=2`; I'll change that in case the issue re-occurs.

SwoopX, 2024-09-21:
This part of your log is pretty remarkable:

```
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | LTG002
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | RWL022
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: dresden elektronik | Kobold
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | LTA009
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: innr | SP 240
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | 915005996501
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: IKEA of Sweden | TRADFRI Driver 30W
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | SML003
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | RDM001
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | LTU001
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: icasa | ICZB-IW11D
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | LCA006
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF identifier pair: Signify Netherlands B.V. | LWA028
Sep 15 12:36:35 pi5 deCONZ[2466075]: 12:36:29:227 DDF try to find raw JSON DDFs for 13 identifier pairs
```

It seems deconz has decided just to load your light DDFs. The rest of the log is not less remarkable, as **all** DDFs are apparently reloaded like at least 10 times in a row without any result. I'm not sure how many different sensor devices are to be expected, but maybe the search/load attempts correspond in this regard?

Anyway, something is seriously going wrong here, but unfortunately the log has not all what could potentially be useful onboard from a debug flag perspective.

ebaauw, 2024-09-21:
>  how many different sensor devices are to be expected

Four, I think: Hue dimmer switch, Hue motion sensor, and Hue wall switch module (2 `modelid` values)

manup, 2024-11-26:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.29.1-beta

This should work now with latest release, and the restart also happening a bit faster.

ebaauw, 2024-12-02:
Seems to be solved, indeed.

closed 2024-12-02 by ebaauw