dresden-elektronik/deconz-rest-plugin #7929: DDF hot reload doesn't re-consider conditions

ebaauw, 2024-09-15:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Got some new LTA009 Hue lights  with ancient firmware, not yet exposing the _Hue Effects_ (0xFC03) cluster.  After firmware upgrade, and  reload of the _Simple Descriptor_, the GUI would show the cluster.  However, _Hot Reload_ doesn't link the DDF to the device.  The DDF would only be matched after deCONZ restart.    Note that the DDF has a filter on the existence of the cluster:  `"matchexpr": "R.hasCluster(0x0b,0xfc03)"`

### Steps to reproduce the behavior

1. Pair LTA009 with old firmware and check that it doesn't expose the _Hue Effects_ cluster;
2. Update the firmware through the OTAU plugin, re-read the _Simple Descriptor_ and notice that the light now exposes _Hue Effects_;
3. Try _Hot Reload_ (on another LTA009 that was already exposed through the DDF), and notice the new LTA009 remains unmatched;
4. Try to select the DDF manually on the new LTA009 and _Hot Reload_, and notice the new LTA009 remains unmatched;
5. Restart deCONZ and notice the new LTA009 is matched to the DDF.

### Expected behavior

Not sure, but I would expect the DDF to be (re-) assigned after updating the device firmware.

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian
- Firmware version: 26780700
- deCONZ version (not Home assistant Addon version!): 2.28.1
- Device: RaspBee II

SwoopX, 2024-09-15:
Interestingly, I had a rather similar experience yesterday while trying to add support for the frient pet aware motion sensor. DDF was there and correctly using name and ID but reading those attributes did not trigger the DDF association mechanism 🤔 Had to delete it in deconz and reset'n're-pair to make it work.

bluemoehre, 2024-10-01:
Run into similar problems - not only using hot reload, but also after restarting deCONZ I had flaky results. 
Deleted devices have still been known by name after re-adding. 
Also old DDF was matched even if device had new firmware and matchexp should not match anymore. 
Additionally having three identical devices: one was not matched by DDF while all others were matched.

More info is here (in case it helps):
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6251

closed 2025-03-15 by github-actions[bot]