dresden-elektronik/deconz-rest-plugin #7931: DDF parsing error if only difference in manufacturername is a space character. Only the first is parsed.

olemr, 2024-09-15:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Found during https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7925
The Sunricher vendor 's Namron manufacturer have released, for the similar HW device 4512729 two Firmware with different manufacturer names. 
Firmware 2.5.3_r20 presents `manufacturername = "NAMRON  AS"`  (note two spaces between N & A)
Firmware 2.5.3_r21 presents `manufacturername = "NAMRON AS"`  (note one space between N & A)

Now, in /usr/share/deConz/devices/sunricher/4button_switch.json DDF, only the first listed NAMRON AS entry is parsed correctly.

With a DDF like:
```
"NAMRON AS",
"NAMRON  AS",
``` 
Switches with r20 FW is not picked up, and Adding fails.
Changing to:
```
"NAMRON  AS",
"NAMRON AS",
``` 
and the Add succeeds.

### Steps to reproduce the behavior

1. Like described in Bug description.

### Expected behavior

Expected DDF parsing to pick up either one of the two NAMRON manufacturer names, not just the first one.

### Screenshots

NA

### Environment

- Host system: Raspberry Pi & PC
- Running method: Raspbian & Ubuntu
- Firmware version: 26400500
- deCONZ version 2.28.1
- Device: ConBee I
- Do you use an USB extension cable: yes + a powered HUB
- Is there any other USB or serial devices connected to the host system? No

### deCONZ Logs

This is from an (unsuccessful) attempt with an r20 switch and the NAMRON with 2 spaces second in the manufacturername list:
```
     45: 21:56:25:217 DDF identifier pair: NAMRON AS | 4512719
  15557: 21:56:25:590 Sqlite group: name = 4512719 38
  15563: 21:56:25:590 DB found group 4512719 38 0xEE01
  15565: 21:56:25:590 Sqlite group: name = 4512719 38
  15571: 21:56:25:590 DB found group 4512719 38 0xEE02
  16555: 21:56:25:630 Sqlite sensors: modelid = 4512719
  17912: 21:56:26:933 DEV modelId: 4512719, 0x60A423FFFE60F3DB
  20490: 21:56:28:038 Websocket 127.0.0.1:36090 send message: {"attr":{"id":"44","lastannounced":null,"lastseen":"2024-09-15T19:56Z","manufacturername":"NAMRON AS","mode":1,"modelid":"4512719","name":"Switch 44 K4 Guestroom","swversion":"2.5.3_r21","type":"ZHASwitch","uniqueid":"60:a4:23:ff:fe:60:f3:db-01-0006"},"e":"changed","id":"44","r":"sensors","t":"event","uniqueid":"60:a4:23:ff:fe:60:f3:db-01-0006"} (ret = 347)
  22691: 21:57:24:854 84:2e:14:ff:fe:90:13:5f/attr/modelid expression: Item.val = Attr.val --> 4512719
  22698: 21:57:24:857 DEV modelId: 4512719, 0x842E14FFFE90135F
  22699: 21:57:24:857 try load DDF from disc for NAMRON  AS -- 4512719
  37647: 21:57:30:005 DEV no DDF for 0x842E14FFFE90135F, modelId: 4512719
```
This is from an (successful) attempt with an r20 switch and the NAMRON with 2 spaces first in the manufacturername list:
[... 9 lines cut ...]
  22990: 22:03:50:661 84:2e:14:ff:fe:90:13:5f/attr/modelid expression: Item.val = Attr.val --> 4512719
  22997: 22:03:50:664 DEV modelId: 4512719, 0x842E14FFFE90135F
  22998: 22:03:50:664 try load DDF from disc for NAMRON  AS -- 4512719
  37972: 22:03:55:758 Websocket 127.0.0.1:49088 send message: {"attr":{"id":"65","lastannounced":null,"lastseen":"2024-09-15T20:03Z","manufacturername":"NAMRON  AS","mode":1,"modelid":"4512719","name":"Switch 65","swversion":null,"type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"},"e":"changed","id":"65","r":"sensors","t":"event","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"} (ret = 328)
  37973: 22:03:55:758 Websocket 192.168.1.2:52517 send message: {"attr":{"id":"65","lastannounced":null,"lastseen":"2024-09-15T20:03Z","manufacturername":"NAMRON  AS","mode":1,"modelid":"4512719","name":"Switch 65","swversion":null,"type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"},"e":"changed","id":"65","r":"sensors","t":"event","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"} (ret = 328)
  37974: 22:03:55:760 Websocket 127.0.0.1:49088 send message: {"e":"added","id":"65","r":"sensors","sensor":{"config":{"battery":0,"group":"20011,20012","on":true,"reachable":null},"ep":1,"etag":"9730a75bbf2b04f2c904c2c7cbf391e0","id":"65","lastannounced":null,"lastseen":"2024-09-15T20:03Z","manufacturername":"NAMRON  AS","mode":1,"modelid":"4512719","name":"Switch 65","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"},"t":"event","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"} (ret = 482)
  37975: 22:03:55:761 Websocket 192.168.1.2:52517 send message: {"e":"added","id":"65","r":"sensors","sensor":{"config":{"battery":0,"group":"20011,20012","on":true,"reachable":null},"ep":1,"etag":"9730a75bbf2b04f2c904c2c7cbf391e0","id":"65","lastannounced":null,"lastseen":"2024-09-15T20:03Z","manufacturername":"NAMRON  AS","mode":1,"modelid":"4512719","name":"Switch 65","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"},"t":"event","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"} (ret = 482)
  38013: 22:03:56:308 [INFO] - Button 1002 - 4512719, broadcast to: 0x6401, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 115
  38018: 22:03:56:716 Websocket 127.0.0.1:49088 send message: {"attr":{"id":"65","lastannounced":null,"lastseen":"2024-09-15T20:03Z","manufacturername":"NAMRON  AS","mode":1,"modelid":"4512719","name":"Switch 65","swversion":"2.5.3_r20","type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"},"e":"changed","id":"65","r":"sensors","t":"event","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"} (ret = 335)
  38019: 22:03:56:716 Websocket 192.168.1.2:52517 send message: {"attr":{"id":"65","lastannounced":null,"lastseen":"2024-09-15T20:03Z","manufacturername":"NAMRON  AS","mode":1,"modelid":"4512719","name":"Switch 65","swversion":"2.5.3_r20","type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"},"e":"changed","id":"65","r":"sensors","t":"event","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"} (ret = 335)
  38022: 22:03:57:017 Websocket 127.0.0.1:49088 send message: {"attr":{"id":"65","lastannounced":null,"lastseen":"2024-09-15T20:03Z","manufacturername":"NAMRON  AS","mode":1,"modelid":"4512719","name":"4512719","swversion":"2.5.3_r20","type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"},"e":"changed","id":"65","r":"sensors","t":"event","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"} (ret = 333)
  38023: 22:03:57:018 Websocket 192.168.1.2:52517 send message: {"attr":{"id":"65","lastannounced":null,"lastseen":"2024-09-15T20:03Z","manufacturername":"NAMRON  AS","mode":1,"modelid":"4512719","name":"4512719","swversion":"2.5.3_r20","type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"},"e":"changed","id":"65","r":"sensors","t":"event","uniqueid":"84:2e:14:ff:fe:90:13:5f-01-0006"} (ret = 333)
  38409: 22:04:34:254 [INFO] - Button 2002 - 4512719, broadcast to: 0x4E2B, endpoint: 0x01, cluster: ONOFF (0x0006), action: Off, payload: None, zclSeq: 118
  38650: 22:04:43:710 DB sql exec REPLACE INTO groups (gid, name, state, mids, devicemembership, lightsequence, hidden, type, class, uniqueid) VALUES ('0xEE01', '4512719 38', 'deleted', '', '', '', 'false', 'LightGroup', 'Other', '60:a4:23:ff:fe:60:f3:db')
  38652: 22:04:43:711 DB sql exec REPLACE INTO groups (gid, name, state, mids, devicemembership, lightsequence, hidden, type, class, uniqueid) VALUES ('0xEE02', '4512719 38', 'deleted', '', '', '', 'false', 'LightGroup', 'Other', '60:a4:23:ff:fe:60:f3:db-1')
  38721: 22:04:43:773 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('65', '4512719', 'ZHASwitch', '4512719', 'NAMRON  AS', '84:2e:14:ff:fe:90:13:5f-01-0006', '2.5.3_r20', '{"buttonevent":2002,"lastupdated":"2024-09-15T20:04:34.255"}', '{"battery":87,"group":"20011,20012","on":true,"reachable":true}', '{"d":1,"ep":1,"in":[0,1,4096],"out":[5,6,8],"p":260}', 'normal', '1', '2024-09-15T20:03Z', '')
  38737: 22:04:44:126 [INFO] - Button 2002 - 4512719, broadcast to: 0x4E2B, endpoint: 0x01, cluster: ONOFF (0x0006), action: Off, payload: None, zclSeq: 119
  39892: 22:05:30:988 Websocket 127.0.0.1:49088 send message: {"attr":{"id":"44","lastannounced":null,"lastseen":"2024-09-15T20:05Z","manufacturername":"NAMRON AS","mode":1,"modelid":"4512719","name":"Switch 44 K4 Guestroom","swversion":"2.5.3_r21" [... cut ...]
  40143: 22:05:43:642 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('65', '4512719', 'ZHASwitch', '4512719', 'NAMRON  AS', '84:2e:14:ff:fe:90:13:5f-01-0006', '2.5.3_r20', '{"buttonevent":2002,"lastupdated":"2024-09-15T20:04:44.127"}', '{"battery":87,"group":"20011,20012","on":true,"reachable":true}', '{"d":1,"ep":1,"in":[0,1,4096],"out":[5,6,8],"p":260}', 'normal', '1', '2024-09-15T20:03Z', '')
```

### Additional context

Tested on both a RPi4 and a Ubuntu 22.04 PC.

SwoopX, 2024-09-15:
As there is no bug involved, I'm closing this.

closed 2024-09-15 by SwoopX

olemr, 2024-09-15:
@SwoopX 
I don't understand. I have described the bug in great detail?
Why do you not consider this a bug?