dresden-elektronik/deconz-rest-plugin #7935: Headless deCONZ - no more sensor updates after searching for lights

jan666, 2024-09-21:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Produkt ConBee III
Gateway Version 2.28.1
Firmware 26500900

I noticed this in the latest stable version before 2.28.1, too: when I search for lights (and add a light, have not tried without adding one) most(?) of the sensors stop sending updates via REST-API/WebSocket. This affects temperature sensors like SONOFFs, open/close sensors from Ikea and also the ZHAPower, ZHAConsomption etc from Ubisys Window Covering devices. So I guess this is more a problem updating the values and not related to zigbee communication? Controlling lights and window covering devices still work.

https://forum.phoscon.de/t/headless-deconz-no-more-sensor-updates-after-searching-for-lights/5120/1

### Steps to reproduce the behavior

Search for lights

### Expected behavior

Sensors should Update After searching for lights 

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / deconz-docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

SwoopX, 2024-09-21:
Was finally able to reproduce that in my production network (may it rest in pieces 🙂 ) with a full blown debug collection. I let it run for some additional minutes and then I'm super curious what is to be found.

SwoopX, 2024-09-21:
So, what I can conclude from my debug log is the following:

Deconz is loading everything as it should on startup. It seems to be key to reproduce the issue that one pairs a device which is not yet known to deconz, meaning there is no cached DDF. When now searching for new devices, it gets paired as it should. When all necessary data is queried from that device, function `&DeviceDescriptions::get(const Resource *resource, DDF_MatchControl match)` is called and subsequently calling `loadDDFAndBundlesFromDisc(resource)`.

With the latter call, things apparently go south when all DDFs are read again. The DDF for the newly added device is working, but everything else ceases working. As if the handles are destroyed or not appropriately refreshed. As the new device is known from then on, this also explains why a deconz restart brings everything back to life: everything is freshly loaded and only once.

sch115, 2024-09-24:
I am having the same issues on Raspberry with Deconz GUI and Conbee II, running latest versions. Sensors such as Temperature and Open/Close stop sending updates even after restarting "Unifi Protect" Child Bridge only. I can also confirm this behaviour after searching for lights as mentioned above.

ebaauw, 2024-09-24:
Saw something similar last weekend (also on a Pi with GUI enabled).  After a Hot Reload of a DDF (for a CO sensor), the light level and temperature reports (and read attribute responses) for my Hue motion sensors were ignored by the API.  They would be reflected in the GUI.  Oddly, the presence reports continued to be handled by the API?

sch115, 2024-09-24:
Yes, everything is still visible and linked in GUI during this issue. And only full restart of the gateway brings it back to work. I also think Hue Wall Modules were impacted too but can't be really sure about this one since at that time I've had no idea the sensors aren't reporting.

mtressl, 2024-09-27:
following...

I see a similar bug /behaviour in my network. Some sensors (Xiaomi) stop reporting but not all of them.

RneeJ, 2024-09-29:
I can confirm the same behavior in my previous install (headless), same application versions. 
For days there was nothing wrong and sensors updated regulary.  
Reporting stopped for some sensors then all stopped. 

Now i have a new install with desktop environment on a RPi 5. 
Also i added 3 extra (mains powered) routers to keep a strong all covering signal.
I keep the deCONZ application up for now. 
Maybe this helps. 

Will report back in some days.

jan666, 2024-10-01:
> Ass soon I delete the new device and restart deConz (GUI here), everything is running fine again.

I dont think you have to delete the new device. Just restart deCONZ. For me just restarting deCONZ fixes this al he time.

ebaauw, 2024-10-01:
You might want to wait some time, for deCONZ to save the resources for the newly paired device to the database, so on deCONZ restart, the DDF can be assigned on startup.  If you restart too soon, the values needed to match the DDF are not known on startup, and the DDFs are reloaded once they're read, causing the issue to reoccur.

ThePhiMuc, 2024-10-01:
Thanks guys! I will give that a try and do some more investigations this evening. 
Maybe I did the restart to soon yesterday. I'll come back to this as soon I got some new insights...

olemr, 2024-10-01:
I don't think waiting is gonna cut it.
When I added a switch during the day, I only noticed when it got dark several hours later that my outdoor lights didn't come on.
HUE outdoor sensor. (My HUE indoor sensor still updated the light level)
After deCONZ restart, update was received 3-5min later.

hyper2910, 2024-10-22:
Any Solution?  A restart works for me, but a SW Solution Sound be Fine

eknirk, 2024-10-24:
> Any Solution? A restart works for me, but a SW Solution Sound be Fine

Try downgrading to 2.26.3. Every later version had the same problem.

fapoo, 2024-11-04:
Absolutely the same issue here. I ordered an aquaria wate leak sensor and as soon I include it, every other sensor (contact, movement, temperature) stops working but the new one. Lights keep working fine though. The first time I didn't notice until two days later so I don't think that waiting will solve the problem. Restart won't solve it neither. I always have to import a backup to make it work again. 
Using 2.28.1. Unfortunately on Home Assistant and I don't know how to downgrade there again. Are the Deconz guys aware of this bug?

bluemoehre, 2024-11-05:
I want to point out: \
**This is a really critical issue**.

A week ago I noticed lights outside not turning off correctly (which I actually didn't bother so much). But since a few days I also had lights not turning on. 

**Imagine someone slips in a winter night …** \
**Even worse if he gets badly hurt …** \
On top, it is proofed there was no light at your house: **You may face a legal case**. 

Also my logs state that some door locks didn't act at night, because daylight was reported for multiple days. This makes me feel very uncomfortable now.

As this bug is hard to see, but at the same time may seriously affect many people without them knowing it, I would like to see this as a top priority.

manup, 2024-11-26:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.29.1-beta

The release is uploaded, would be cool if anyone could check if the updates now are processed normally.

Darkem1, 2024-11-27:
Unfortunately nothing changed, the issue is still present.

snozzlebert, 2024-11-28:
I also experience similar issues. All these issues are probably related to DDF (re)loading.
I reverted to 2.27.5 which has fewer issues compared to 2.29.1 (only issues after searching for sensors/lights like described in this issue). 
For all 2.28 and 2.29 versions my Hue dimmer switch (old and new), Hue wall switch module and some Xiaomi contact sensors stop sending events after a while (few hours after start). Similar as mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7927 only I don't do restarts via API.

bluemoehre, 2024-11-28:
@manup please re-open this issue - it doest not seem to be fixed.

ebaauw, 2024-11-28:
I can confirm: After adding a new device in v2.29.1, the Hue dimmer switch `sensors` resource is no longer updated, even though the deCONZ GUI registers the FC00 command alright.  After deCONZ restart, everything seems to be working again.

manup, 2024-12-02:
> I can confirm: After adding a new device in v2.29.1, the Hue dimmer switch `sensors` resource is no longer updated, even though the deCONZ GUI registers the FC00 command alright. After deCONZ restart, everything seems to be working again.

Is this with the vanilla installed deCONZ? Part of the fix is in deconz-lib so updating the REST plugin repository alone isn't enough.

I've just re-tested by freshly joining a mains powered plug and a Xiaomi weather sensor. Here the Hue switches keep operating normally. I'll try to test this in another setup.

bluemoehre, 2024-12-02:
Is this some kind of memory leak thing? I can clearly see that over time more and more nodes start acting weird or stop responding completely. To me it takes several days to weeks in my Conbee 2 mesh; but I can force problems by adding new lights.
Maybe it's related to the network size? Since I do not have such problems in my second network running Raspbee 2 with only 2 sensors. 
Degradation starts with the (non router) sensor nodes, then somewhen individual lights stop responding to group commands. Later on I could no longer switch single lights on and off.

manup, 2024-12-02:
How large is your network? My production network has approx. 50 devices, also sometimes running for 1-3 weeks but don't see hiccups here.

I don't think it's a memory leak. But you have a good point which I'll try to recreate, by joining new lights. So far I only did this with one Ledvance plug, perhaps the problem comes up when joining more lights of various brands.

bluemoehre, 2024-12-02:
> How large is your network? 

Is there a counter in deCONZ somewhere? Approx 80 nodes atm. 

My unordered observations and guesses based on issues I had myself or read somewhere the past years: \
Some DDFs have too short polling intervals (eg. IKEA); the communication log often scrolls very rapidly cuz of state reports, but nothing was changed; sometimes commands start getting queued, timeouts stack up, devices are marked as zombies, commands get dropped/lost;  adding new lights causes other devices to drop out; feels like an exponential problem related to the amount of (some specific) devices or the mix of passive/non-router/battery vs others.

Just a short summary of what I see a lot scrolling through the log, when things go bananas:
- `ZCL attribute report`
- `force poll`
- `state change failed`
- `delay sending request`
- `5 running tasks, wait`

I have the feeling there are bugs in some queues or in some garbage collection. Everything goes back to normal after restart - mostly.

ebaauw, 2024-12-02:
> Is this with the vanilla installed deCONZ? Part of the fix is in deconz-lib so updating the REST plugin repository alone isn't enough.

I installed v2.29.1 through `apt`, then re-compiled the REST API plugin with merged changes from upstream/master.

> How large is your network? My production network has approx. 50 devices, also sometimes running for 1-3 weeks but don't see hiccups here.

In this case: 92 devices.  It's actually a new installation with a ConBee III, since I couldn't get the RaspBee II to change Zigbee channels.  Now channel 25, solving my interference issues.

>  But you have a good point which I'll try to recreate, by joining new lights.

I think the issue happens when joining a device which will be exposed by DDF.  Not sure if it's 100% reproduceable.

> I don't think it's a memory leak.

I don't think either.  When not changing the network (joining new devices), deCONZ runs very stable.  I have very clear symptoms: deCONZ updates from the device (_Report Attributes_ of _Read Attributes Response_), updates the GUI, but doesn't update the REST API items.

> Just a short summary of what I see a lot scrolling through the log, when things go bananas:

I think that's a different issue.  Those messages seem related to radio interference issues, which, in my experience, will eventually fill up some queues (in the core or in the coordinator device firmware), from which deCONZ eventually doesn't recover.  Sometimes I even needed to reset (power cycle, or GCF_Flasher -r) the coordinator device to clear the condition.  Once things go "bananas" this way, devices (mostly end devices) drop off the network (presumably since they no longer receive ACKs on their reports).  Sometimes they recover (Hue devices tend to, eventually), sometimes, indeed, they need to be re-joined.

>  individual lights stop responding to group commands

I've seen several times that the group command queue in the coordinator device would fill, but then none of the lights would react anymore.  Barring a bug in the light firmware, this points to radio interference.

Idaho947, 2025-01-16:
Hello, any news about this issue ?

bluemoehre, 2025-01-18:
Hey folks, watching this topic... 
I have the feeling, Dresden is not caring (much) anymore. 

Just check the Github repo yourself:

- Issues are mostly closed by the bot due to inactivity
- Bugs remain for multiple months
- Device requests remain for multiple months
- Pull Request are piling up for months to years 

At least it's very bad communication by the company (not the community and moderators). 
**Maybe we should create another Issue asking for the status of the development & project & capacities involved.**

Because of exactly this issue here and similar ones, which are very crucial and significant for the smart home stability, I migrated away last year. I simply was not willing to wait any further for major bugfixes, being stucked with trouble at my home about the sensors and lighting. I personally will still care about the issues I opened or participated (that's why I am also watching this one). 
But that's it for now. I feel very sorry for everyone here investing lot's of time. 

> [!NOTE] 
> **Who wants to join a discussion** about the project status, please **just leave an emoji** on this comment, so we can shift it to another place if there's enough interest.

manup, 2025-01-18:
Hey guys, very sorry for the large delay in response. The last months have been difficult time wise but you're right I should have communicated this. We're catching up now to get the project back on normal release schedule and importantly focus on fixing the harder bugs.

I'll post a rough road map on whats to be worked on / fixed currently and for the near future in the next days and where we're heading with deCONZ in 2025.

Mimiix, 2025-01-19:
@manup Please do these roadmap related updates on the forums or in another place :) Just to keep this on topic again!

bluemoehre, 2025-01-22:
> [@manup](https://github.com/manup) Please do these roadmap related updates on the forums or in another place :) Just to keep this on topic again!

Why not add a pinned note at the [Wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki) frontpage or use the [Milestone Feature](https://docs.github.com/en/issues/using-labels-and-milestones-to-track-work/about-milestones)?

I suppose most people hit the Github repo first, and the milestones are also primarily code-related. Does that make sense? =)

Mimiix, 2025-01-23:
Hi all,

Manup just wrote the roadmap. You can find it here https://forum.phoscon.de/t/2025-deconz-development-overview-roadmap-ish/6088/1

jan666, 2025-02-05:
I got a [Ubisys R0](https://www.ubisys.de/produkte/produkte-zusatzoptionen/produkte-zusatzoptionen-r0/) recently (with Zigbee and bluetooth). I integrated it and used the Ubisys Smart Home App on my iPhone to access my Zigbee Network via Bluetooth. This all works but I dont get any sensor updates as long as the R0 is connected to my network. When I power it off and restart deCONZ everything works again. When I power it on again - no more sensor updates.

Is this related or another bug?

jan666, 2025-03-12:
Is there anyone working on this? The issue is almost 6 month old

Idaho947, 2025-04-10:
Any news about this issue ? Fix in the new beta ? @Mimiix ?

Idaho947, 2025-04-10:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7935#issuecomment-2506784798

I see that comment several mounth ago and think it s not. But maybe miss something.

snozzlebert, 2025-04-12:
No 2.29.1 did not fixed it, see my comment earlier: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7935#issuecomment-2506680250
2.29.3 neither.
I did an upgrade today to 2.29.3 but it became a mess. Again all Aqara door sensors and Hue wall switch module stopped working. Reverted back to 2.27.5. I've bought another zigbee stick and I'm planning a migration to zigbee2mqtt. I rather not want to, because I've to repair 70 devices. But the deCONZ software became too unstable for me (while firmware and hardware are ok). I'm almost only using Hue lights and sensors and Aqara door and temperature sensors. Never had issues before all this new DDF stuff :(

Jagohu, 2025-04-14:
Same here. I moved to 2.28.1 a month ago, issues started to happen. Switches not seen, lights not showing up, sensors not updated... I bought a Connbee III, flashed it to latest firmware, tried 2.29.1, 2.29.2 - no joy.

Now I downgraded to 2.26.3 but even there I can't add a Philips indoor motion sensor which was working flawlessly before...

manup, 2025-04-14:
Hi can you please try with the newly released ConBee III firmware https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeIII_0x26550900.bin.GCF ([Changelog](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog)).

It contains fixes for joining/rejoining and keeping devices in the network. We've testes with various devices like the hue dimmer and other sleeping and devices. Feedback would be very welcome.

I'm also currently extending my network with more sleeping devices to check in the sniffer if there are more odds to fix. I'm preparing a few smaller PRs to get rid of unneeded traffic towards devices which can amplify problems.

Jagohu, 2025-04-14:
[image]
Yep, this is what I have

Jagohu, 2025-04-14:
I just tried again with two things
1) Philips Hue indoor motion sensor (the one with the button, no pin needed, I believe SML003 showing up as Signify Netherlands B.V. - nothing at all. I have another 4 of them running without issues.
2) Xiaomi wall switch QBKG11LM - the switch is created when seaching for switches but the light/plug not.
Correctly recognized as all others - I have a few of them but for this one (last one) no light/plug even with 2.26.3. It was created once with 2.28.1 then I had to change Zigbee channel from the default 11 to 25 and then the light/plug disappeared and doesn't come back anymore. Same story with the motion sensor.

`"252": {
        "config": {
            "on": true,
            "reachable": true,
            "temperature": 2100
        },
        "ep": 5,
        "etag": "d5df765e16aa4738b3dfd099f42db99b",
        "lastannounced": null,
        "lastseen": "2025-04-14T10:37Z",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.ctrl_ln1.aq1",
        "name": "GarageBack",
        "state": {
            "buttonevent": 1002,
            "lastupdated": "2025-04-14T10:37:32.568"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:04:17:3f:ea-05-0012"
    },`

manup, 2025-04-14:
Which Operating System are you running? I'd like to focus on the SML003 first to get it working. My upcoming PRs reduce traffic to Philips sleeping devices, in my setup these worked fine for a while but it's better to reduce it anyway.

Would be good if we can test a pre-build in your setup and perhaps check the logs and sniffer where it goes wrong.

Jagohu, 2025-04-14:
Raspbian Bullseye

I'm running some logs while running deconz from command line with _/usr/bin/deCONZ -platform minimal --auto-connect=1 --dev-test-managed=1 --dbg-aps=2 --dbg-info=2 --dbg-error=2 > /var/log/deconz1.log_

Even when I press setup on the Hue sensor its ID doesn't show up in the logs... I thought with these options any joiner would be ok.

manup, 2025-04-14:
Ok lets see if we can get the device on track with v2.29.3 and firmware version we should see at least a Device Announcement when joining the sensor. This is by openening the network via *deCONZ > Panels > Control > Pairing > Enable permit join* (or starting sensor search in Phoscon App.

Then pressing the setup button for 10-15 seconds until the sensor LED blinks, a few seconds later the log should show messages with "device announce". This would be the initial stage to join the device before configuration.

Can you see that in the logs?

Jagohu, 2025-04-14:
I apologize in advance for probably doing something stupid.

I'm running on a headless RPI3. So I stopped deconz via systemctl, vnc-d into the RPI and tried to start it via the Deconz icon - no joy, the icon doesn't seem to do anything.
I only managed to start in a few times in the past years, even then I could only see my own Connbee and not much more. :(

Tried to start it from lxterminal via the .sh file but it tells me to start it from the systemd service. When I do and try again it says it's already running.

After restarting deconz as a service, I tried adding the sensor on 2.29.3 via the webui but it still doesn't show up though.

manup, 2025-04-14:
When starting with your command line from above:

```
/usr/bin/deCONZ -platform minimal --auto-connect=1 --dev-test-managed=1 --dbg-aps=2 --dbg-info=2 --dbg-error=2 > /var/log/deconz1.log
```

The log should be found in `/var/log/deconz1.log`.

Your Raspbian Bullseye is 32-bit or 64-bit? I'd like to build a version later on.

Idaho947, 2025-04-14:
Just flash with the last firmware and all my zigbee networe is down. See the key but no link with my element...

Idaho947, 2025-04-14:
Hello, I let it run 10 min. Can do the test again it's my pro netwok but send the zll.db file. Regards

manup, 2025-04-14:
Thanks I'll have a look, my assumption is that way to soon requests to the sleepers are filling up the queue. My PRs from earlier today should help for this — so for the test it might be good to wait a bit until the build with these fixes is ready. There's more that can be done, I test some ideas now.

Jagohu, 2025-04-14:
So I've left for work and let it run for the past 7 hours.

My Hue motion sensors found their way back for the presence/temperature part but the Lightsensor part is not updated.
Open/close, temperature, smoke sensors are not updated, even though they show up with a black background in Phoscon.

Jagohu, 2025-04-14:
64bit.
Following a dist-upgrade (after reading this https://forum.phoscon.de/t/phillips-hue-motion-sensor-sml001-doesnt-update-brightness/5815)  and a restart things start to find their way to the coordinator, lights are looking ok too on the Philips ones. I'll see by the morning,

jan666, 2025-04-15:
Does this also finally fix the bug that is this issue about?

Jagohu, 2025-04-15:
On the first look yes.
My sensors made it back to "reachable" but I need to re-pair many of them in the end.

I tried to add lights/plugs and my sensors seem to be updated following it. I'll keep testing it.

I still have the issue with that lumi.ctrl_ln1.aq1 swich though which only shows up as a switch and not as a light/plug. Weird. There are quite a few others of the same type, working perfectly.

closed 2025-04-15 by manup

manup, 2025-04-15:
I don't think it's the firmware, with 0x26550900 there aren't known problems yet. We need to check out the logs with deCONZ v2.29.4-beta and firmware version 0x26550900 to see status codes of commands send out and received as well if deCONZ/plugin is queuing up somewhere.

Idaho947, 2025-04-16:
@manup Could this be an ‘incompatibility’ between 2.28.1 and the new firmware? Because that's all I've changed.

2.28.1 + 0x26530900 = OK
2.28.1 + 0x26550900 + reboot = KO
2.28.1 + reflash 0x26530900 + reboot = OK

Can't make test before the end of the month.

jan666, 2025-04-16:
any ETA for a new stable release?

jan666, 2025-04-30:
Just updated deCONZ

2.29.5
26500900

and it looks good so far! Also fixes my problem with the Ubisys R0. 

Thanks!