dresden-elektronik/deconz-rest-plugin #7938: Fire Angel AngelEye Koolmonoxidemelder Zigbee Smart Home ZBCO-AE-10X-EUR

ebaauw, 2024-09-22:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

AngelEye Koolmonoxidemelder Zigbee Smart Home

### Manufacturer

Fireangel

### Model identifier

Alarm_SD_Device

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

#### Poll Control
[image]

### IAS Zone
[image]

### IAS WD
[image]

ebaauw, 2024-09-22:
I got the sensor at the local DYI store, https://www.karwei.nl/assortiment/angeleye-koolmonoxidemelder-zigbee-smart-home-10-jaar-batterij/p/B131917.  The sensor itself contains a non-replaceable battery that should last 10 years (which is probably the life time of the sensor hardware as well).  It contains a detachable Zigbee module, with a replaceable CR2 battery.  According to module's manual, it can be used with other sensors as well, see also #6111 and #6545.

I just paired the device with deCONZ and wrote the coordinator's MAC address to the _IAS_CIE_Address_ attribute in the _IAS Zone_ cluster.  I seem to remember, deCONZ would assign _Zone ID_ 100, but it reports 0.  The legacy code exposes the _IAS WD_ cluster is exposed as a `lights` resource; the _IAS Zone_ isn't exposed as `sensors` resource, since the device is not whitelisted.
```json
{
  "capabilities": {
    "alerts": [
      "none",
      "select",
      "lselect"
    ]
  },
  "config": {
    "groups": [
      "0"
    ]
  },
  "etag": "662ccd3b9fa9b237c3b04710bd561a50",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2024-09-22T15:36Z",
  "manufacturername": "Fireangel",
  "modelid": "Alarm_SD_Device",
  "name": "Laundry Room CO",
  "state": {
    "alert": "none",
    "reachable": true
  },
  "swversion": "20201218",
  "type": "Warning device",
  "uniqueid": "00:15:5f:00:dc:ba:2f:83-01"
}
```

It might be challenging to create a DDF for this device, as the same Zigbee module might be used for different types of sensors.  We would need to check the _Zone Type_ to know what `sensors` resource type to create, but that's only populated after enrolling the device, which would only happen once a DDF has been matched.  I'm tempted to whitelist the device in the legacy code.  I think that should handle the different sensor types without any issue.  @manup @SwoopX  any ideas here?

The device is a light sleeper, polling every 7 seconds; I didn't change any _Poll Control_ settings.

The _Zone Status_ indicates the device sends _Restore reports_, but no _Supervision reports_.  Need to see if it accepts attribute reporting for periodic reports.

The siren can be activated through the _IAS WD_ cluster (yeah!), but I only managed to do that using the _Squawk_ command.  Not sure if it doesn't support _Start Warning_, or if I simply haven't yet found the magic combination of parameters.  In the first case, we need some hack to activate the siren from the API, as setting `"alert"` normally sends a _Start Warning_.

I suspect the _Power Configuration_ cluster reports the replaceable battery of the Zigbee module.

SwoopX, 2024-09-22:
That device is interesting and indeed tricky to handle. If the zone type is just populated after enrollment, I'm not sure if we could even nail that down with legacy code? That somehow feels like a major show stopper 🤔 

Just as a side note: I made the alarm item fit for DDF, so you can assign whatever command you like. Check out the Develco smoke detector.

ebaauw, 2024-09-24:
> if the zone type is just populated after enrollment

I stand corrected: it is populated alright, even before enrollment.  Got a fire sensor as well, just to test the differences.  Other than the _Zone Type_, I see no differences is how the Zigbee module exposes itself.  So if we can somehow use that attribute in a `matchexpr` we're OK.

> Just as a side note: I made the alarm item fit for DDF, so you can assign whatever command you like. Check out the Develco smoke detector.

Brilliant.  I'll use that to send _Sqawk_.

closed 2024-10-09 by manup