dresden-elektronik/deconz-rest-plugin #7942: Costum ESP32-C6 Switch

dark-squirrel, 2024-09-25:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

ESP32-C6

### Manufacturer

Espressif

### Model identifier

C6

### Device type to add

Switch

### Node info

- Node info
[image]

### Endpoints and clusters

- Endpoint and Clusters
[image]

### Basic

- Basic
[image]

### Further relevant clusters

Hello, I have a costum ESP-32 switch and would like to integrate it into Deconz. 
Thanks in advance.

ebaauw, 2024-09-25:
Is this a wireless (controlling smart devices) or a dumb switch (smartifying dumb devices)?  From the fingerprint, I would guess the first.  What _On/Off_ commands does it send when the switch button is pressed?  _On_, _Off_, _Toggle_?  Does it also send _Scenes_ commands, or only _On/Off_.

Did you read the attributes of the _Basic_ cluster?  The switch needs to expose values for _Manufacturer Name_ and _Model Identifier_ to be recognised (and exposed) by the REST API.  I'm guessing you can define these somewhere in the switch firmware.

dark-squirrel, 2024-09-25:
This is only a dumb switch device,you can press a button and it sends a Toogle On/Off. 
And the Attributes are not correct, here are the correct values:
- ZCL Version: 8
- Manufacturer Name: ESPRESSIF
- Model Identifier: esp32c6
- Power Source: Unknown
- Cluster Revision: 4