dresden-elektronik/deconz-rest-plugin #7944: Deconz doesn't load DDFs

TermnX, 2024-09-25:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have a Raspberry pi running the Phoscon desktop image with a Raspbee2 and somehow deconz(2.28.1) doesn't load any DDFs. Not for the RaspBee, which is marked as 0x0000 and witout attributes and not for my Aqara WSDCGQ11LM temperature sensors. When i try to assign the DDF manually it tells me "failed to open <filename.json>"

### Steps to reproduce the behavior

Not sure if reproducable as it may be due to some error with my device or a faulty imaging process. However it may also be because i start i direcly with the deCONZ execuable because the home menu shortcut doesn't work too 

### Expected behavior

Usually deconz should load DDFs automatically or atleast be able to open them from their native folders?

### Screenshots

[image]
[image]
[image]

### Environment

- Host system: Raspberry Pi 4
- Running method: Raspbian
- Firmware version: 
- deCONZ version (not Home assistant Addon version!): (2.28.01)
- Device:  RaspBee II
- Is there any other USB or serial devices connected to the host system? No

### Additional context

Only the Raspbee as a Coordinator and the goal is to have several of the aqara sensors

Mimiix, 2024-09-25:
Hi,

The missing + shows that the devices are not paired correctly/fully paired.  That probably is the reason for the DDF not loading.

What devices are this?

ebaauw, 2024-09-25:
deCONZ hasn't yet read the endpoints nor simple descriptors of your devices, so technically it doesn't even know if they have a _Basic_ cluster, where it can retrieve the manufacturer and model to match the appropriate DDF.  Right click on the node and select _Read Node Descriptor_, _Read Active Endpoints_ and _Read Simple Descriptors_.  A plus sign should appear in each node, as the simple descriptors are read (cf. the Raspbee node).  You probably need to wake the sensors for the reads to succeed.  Then search for new devices, press the + and select the _Basic_ cluster and read its attributes in the _Cluster Info_ panel.

TermnX, 2024-09-26:
@Mimiix these are (Xiaomi) Aqara WSDCGQ11LM temperature, humidity and pressure sensors. The way @ebaauw described for connecting the sensors fully somehow won't work for me, i just get a red blinking 'status LED' on the corresponding node (to indicate failure of these requests i guess?)

Mimiix, 2024-09-26:
This seemingly is not a bug but rather something else. 

Therefore I ask you to ask this question to the forums and figure out there

closed 2024-09-26 by Mimiix