dresden-elektronik/deconz-rest-plugin #7947: Hue wall switch new model RDM004 does support bindings

ebaauw, 2024-09-28:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Unlike the original model Hue wall switch (RDM001), the new model (RDM004) does support hybrid mode, sending both button event notifications from the (server) _Hue Button_ cluster (0xFC00) and _On/Off_ commands from (client) _On/Off_ cluster.  As usual, the Hue-specific _Configuration_ attribute (0x0031) on the _Basic_ cluster needs to be set to 0x000B for this.  Both unicast and group bindings from the (client) _On/Off_ cluster seem to work.

There is one caveat, however: the _On/Off_ commands are sent for both inputs, when configured as _Dual Rocker Switch_.

Note to self: need to check what commands are sent when the wall switch module is configured for push buttons.

### Steps to reproduce the behavior

Write 0x000B to _Configuration_ in the _Basic_ cluster, and create a binding from the (client) _On/Off_ cluster to a light or group.  Observe that the light state changes and the FC00 button event is received.

### Expected behavior

API should expose `config/groups` on the ZHASwitch `sensors` resource.

### Environment

- Host system: All
- Running method: All
- Firmware version: All
- deCONZ version (not Home assistant Addon version!): v2.28.1 and earlier
- Device: All
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

### deCONZ Logs

```
Sep 28 12:16:33 pi5 deCONZ[1825028]: 12:16:33:205 [INFO] - No button map for: RDM004, broadcast to: 0x006E, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF_WITH_EFFECT (0x40), payload: 0000, zclSeq: 34
Sep 28 12:16:33 pi5 deCONZ[1825028]: 12:16:33:248 [INFO] - No button map for: RDM004, unicast to: 0x0000, endpoint: 0x01, cluster: 0xFC00, command: 0x00, payload: 0100003000210000, zclSeq: 33
```

pimw1, 2024-10-15:
Hi Erik, what does the hybrid mode mean in practice? Does it mean that the switch can directly switch lights on/off without the use of the zigbee coordinator?
I've the old model (RDM001) and i'm wondering whether an upgrade would be worth it.

ebaauw, 2024-10-15:
Short answer: yes.  Long answer follows:

Normally, wireless switches/buttons/dials/dimmers either work as controller or as sensor.
- Most non-Hue switches work as controller, sending commands directly to bound devices or a bound group (using **client** _On/Off_, _Level Control_, _Scenes_, or _Color Control_ clusters).  These switches continue to control lights, when the gateway is unavailable.  
For these devices, deCONZ eavesdrops on the group commands, to translate the commands to `buttonevent` values (through the `button_map.json`), so they can be used in deCONZ `rules`.  Typically the API exposes `config.group` on these switches, to setup the binding from the switch to the group;
- Hue switches (and some others) work as sensor, reporting the `buttonevent` values to the gateway (using the **server** FC03 cluster).  That means you need the gateway to be operational for these switches to control lights, through `rules`.

Some Hue devices (like the Hue dimmer switch, the 2nd gen Hue motion sensors and, as we recently figured out, the 2nd gen Hue wall switch module) can be working in (what I like to call) "hybrid" mode, simultaneously as sensor and as controller.  However, we don't (yet?) expose `config.group` on all these devices (as we don't want to pollute the Zigbee network with unused group messages), so you would need to setup the binding manually (in the GUI).  Also, not the limitations of the 2nd gen Hue wall switch module: no dimming (even in push button mode), and only one group (even when in dual mode).

Note that, strictly speaking, the coordinator (the RaspBee or ConBee device) plays no role here.  It's the deCONZ software (or more precise: the REST API plugin) that acts as gateway between the Zigbee and IP networks and executes the `rules`.

pimw1, 2024-10-15:
Got ya, thanks. In my current setup with 2x RDM001 combined with something like this: https://www.girastore.nl/314700.html

So i've basicaly a left part (up and down buttons) and a right part (up and down buttons). Left side is connected to a RDM001, right side is connected to another RDM001.

In my current setup:
- pushing up -> lights on (when currently off)
- pushing up -> next scene (when currently on)
- pushing down -> lights off
- holding up -> increase brightness
- holding down -> decrease brightness
This is implemented via the sensor method of the RDM001 -> Deconz -> Home Assistant automation -> Deconz -> light controll

Ideally, i would like to make advante of the hybrid approach to make the system more robust. I would like to keep it working exactly as described above, unless Deconz/Home Assistant are not responding for whatever reason. In that case, i would like it fall back to the controller mode:
-  pushing up -> lights on
- pushing down -> lights off

Reading your explanation, i don't think this is possible because the distinction between "lights on" (when lights are off) versus "next scene"  (when lights are on) cannot be made?