dresden-elektronik/deconz-rest-plugin #7948: IKEA INSPELNING Smart Plug with kWh meter

NotSoScientific, 2024-09-28:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

INSPELNING

### Manufacturer

IKEA

### Model identifier

INSPELNING Smart plug

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control
[image]

### Color Control

### Thermostat

### Simple Metering
[image]

### Electrical Measurement
[image]
[image]
[image]
[image]
[image]

### Any other cluster of relevance/interest

### OTAU table row

Reports manufacturer 0x117C and image 0x9f3e
[image]

The device currently exposes itself to the REST API as an outlet, which allows for the on/off control, however the kWh meter seems to not report yet. 

This IKEA Zigbee outlet is currently rolling out globally. The IKEA subreddit has reported that an OTA Update may be needed to use all features, this device was not seen in the OTAU file folder. If you have further questions, feel free to reach out.

bluemoehre, 2024-10-01:
Interesting device! Especially after the new Firmware (v45) for Xiaomi / Aqara Plug seems to remove some features like Voltage, Current, Status after Powerloss.

**Before integrating a DDF**,
- ensure Electrical Metering (`0b04`) multipliers `0x0604` & divisors `0x0605` are handled correctly
- ensure Simple Metering (`0702`) multipliers `0x0301` & divisors `0x0302` are handled correctly

... which both currently is still an issue for other products and causes devices to report wrong values.

Additionally it's a real struggle to fix this locally by custom DDFs due to another problem in merging general definitions over custom ones.

**Related files:**
[/devices/generic/items/state_power_item.json](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2b890eeebe6d998331415adbb955bb7cb8956022/devices/generic/items/state_power_item.json)

ebaauw, 2024-10-01:
That won’t work, for a number of reasons:
- The API standardises on weird units (like 0.1A, if memory serves).  This was needed in the time it couldn’t handle floating point values;
- You would need to apply both the multiplier and divisor, but only when supported by the device (or default them to 1);
- Historically, not all devices report correct multiplier and/or divisor values.

it’s probably easiest to hard- code the conversion per device model in the DDF.  What problem merging general definitions are you referring to?  I find this works without any issues - the DDF overwrites any generic item.

bluemoehre, 2024-10-01:
> You would need to apply both the multiplier and divisor, but only when supported by the device (or default them to 1)

That's correct, I simply didn't add this to my example. But as far as I see the specs - the values default to `1`!

> The API standardises on weird units (like 0.1A, if memory serves). This was needed in the time it couldn’t handle floating point values

> Historically, not all devices report correct multiplier and/or divisor values.

Okay, got that one. But doesn't it make more sense to make it better now, make it more generic & become more future proof, and just care about the legacy with custom rules for the old devices, rather than making weird moves for everything? 

> What problem merging general definitions are you referring to?

Gonna make a dedicated ticket soon, so we don't mix up things in this one.

bluemoehre, 2024-10-07:
FYI: IKEA also offers another smart plug and a bundle with a RODRET, both called TRETAKT. That plug is visually identical besides having a white front. **TRETAKT does not support any metering**, it's a simple on/off device.

janip81, 2024-10-08:
I tried the DDF made by @bluemoehre and it works i get measurments but should it be so slow on updating? i get power updates like every 5minutes only, could this be configured to be more precise?

Mimiix, 2024-10-08:
> I tried the DDF made by @bluemoehre and it works i get measurments but should it be so slow on updating? i get power updates like every 5minutes only, could this be configured to be more precise?

Yes, report intervall in deCONZ. But this is not the place to ask ;)

ebaauw, 2024-10-08:
The attribute reporting configuration is standard Zigbee functionality.  Double-click on the attribute in the GUI for a popup window.  These can be set in the DDF as well, under the bindings section.

Note that this configures the reporting interval, not necessarily the interval at which the device takes a measurement.

bluemoehre, 2024-10-08:
> These can be set in the DDF as well, under the bindings section.

This is the part I didn't understand yet - sorry. I found you can drag & drop attributes to the DDF editor's binding section, but fields I see for other devices, do not appear here. Also this doesn't add a reporting property to the DDF.
Changed settings in the modal (after double-clicking an attribute) are not reflected in the DDF as well, so I don't know how to get these settings copy pasted there.

> Look in the GUI user guide under help.

Unfortunately it's not possible to use the GUI help since I switched to Conbee II. If you click it, nothing happens. But that's another issue. The docs I was checking are the official ones linked in the README: 
https://dresden-elektronik.github.io/deconz-dev-doc/modules/ddf/
But these do not explain that specific part of the bindings/reporting (yet).

Anyhow, I guess it would make sense to setup some faster reporting settings by default. 
Is there some default value for all plugs or similar?
And is there a copy of the help files outside of deCONZ GUI, so I can read it?

ebaauw, 2024-10-08:
I don't use the DDF editor that often, but under the _Bindings_ tab, you can setup bindings and attribute reporting, by dragging the cluster into the left area _Bindings_ and subsequently dragging the attribute into the right area _Reporting Configuration_, while the binding is selected.

> If you click it, nothing happens

Same for me, so that's probably an omission in the latest version.

A quick Google search for "deCONZ User Manual" finds https://www.dresden-elektronik.com/wireless/software/deconz.html?file=files/dresden-elektronik/content/downloads/dokumente/funktechnik/deCONZ-BHB-en.pdf.  See sections 7.2.3 and 8.1 for how to setup attribute reporting and bindings.

bluemoehre, 2024-10-08:
> [...] and subsequently dragging the attribute into the right area Reporting Configuration, while the binding is selected.

🤯 I would have never guessed it works that way. That was the missing part for me!

> See sections 7.2.3 and 8.1 for how to setup attribute reporting and bindings.

Actually I was on that, but is it up-to-date? It looks different than the current UI and I didn't find what you told me above ... or it's too unclear / confusing to me.
(In my opinion the linked docs in the README should enable every dev to do daily tasks, so we get a better / faster flow, involving less people / time. Especially adding new devices seems to be a very frequent job. But again - that's another topic)

**Many thanks**, @ebaauw ! I gonna try adjusting the DDF again and check if the device will report in a more meaningful way.

bluemoehre, 2024-10-08:
> i get power updates like every 5minutes only, could this be configured to be more precise?

@janip81 please try the updated DDF. It now uses more meaningful report intervals:

- **on state**: every 1s - 10min
- **electrical measurements**: every 10sec - 10min
- **consumption**: every 1min - 1hour

janip81, 2024-10-09:
> @janip81 please try the updated DDF. It now uses more meaningful report intervals:
> 
> * **on state**: every 1s - 10min
> * **electrical measurements**: every 10sec - 5min
> * **simple metering**: every 1min - 1hour

I seems to get faster usage (Watt) reading now when power changes. so seems to work. Will do more testing

ebaauw, 2024-10-09:
Please make sure the refresh rate for reading an attribute is 60s longer than the periodic reporting rate for that attribute.  We only want to fall back to polling the device, in case reports go missing.

bluemoehre, 2024-10-09:
Should it be exactly `60s`, or just higher than the max reporting interval? Currently it is at the double or multiple. Is there a best practice or standard for this?

NotSoScientific, 2024-10-10:
#7959 allows the Consumption and Power to be reported throught REST. However, the other variables, such as Voltage, are not (yet) directly reported. Thanks everyone for your contributions!

bluemoehre, 2024-10-10:
@NotSoScientific here voltage is reported using that DDF. Can you re-check and maybe re-join the device?

[image]

NotSoScientific, 2024-10-10:
@bluemoehre  

Yeah I have the same in the Phoscon app:
[image]

However, when accessing throught REST in Home Assistant, it shows the following entities:

[image]

This is after refreshing, I have not had the time yet to readopt the device, will update this weekend.

bluemoehre, 2024-10-10:
You can also try in HA to reload deCONZ device list first. So you won't have to re-add it to the network.
Goto `http://\<homeassistant:port\>/config/integrations/integration/deconz` and hit "reload".

ebaauw, 2024-11-07:
I got an INSPELNING (firmware 2.4.34) and installed it for the kitchen boiler (rated 2200W).

To make sure I wouldn't miss any reading, I setup very aggressive attribute reporting (1/300/1) for _RMS Voltage_, _RMS Current_, _Active Power_, _AC Power Multiplier_ and _AC Power Divisor_.

I'm reading the following:

Boiler | raw voltage | m/d | actual voltage | raw current | m/d | actual current | raw power | m/d | actual power
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
heating | 229 | 1/10 | 229V | 9148 | 1/1000 | 9.148A | 2035 | 1/10 | 2035W
standby | 229 | 1/10 | 229V | 8 | 1/1000 | 0.008A | 4 | 1/10 | 4W? 0.4W?

From these readings, I conclude: 
- The reported voltage m/d is wrong; it should be 1/1 instead of 1/10;
- The reported current m/d seems right: the rated 2200 W would correspond to a maximum current of 9.6A (at 229V);
- The reported power m/d is wrong; it should be 1/1 instead of 1/10.  With 9.148A at 229V, the power would be 2095W.

Note that I've never had a smart plug where power = voltage x current holds exactly.  Also different smart plugs report different voltage, even when on the same power line.

While the m/d attributes are reportable, I don't see their values change.

If I read https://github.com/Koenkk/zigbee2mqtt/issues/23961 correctly, the m/d attributes would be fixed in firmware 2.4.45.  Upgrading now...

ebaauw, 2024-11-07:
Ok, with 2.4.45, I'm reading:
Boiler | raw voltage | m/d | actual voltage | raw current | m/d | actual current | raw power | m/d | actual power
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
heating | 2287 | 1/10 | 228.7V | 9218 | 1/1000 | 9.218A | 2048 | 1/1 | 2048W
standby | 2287 | 1/10 | 228.7V | 8 | 1/1000 | 0.008A | 4 | 1/10 | 0.4W

So the m/d attributes are now correct.  Some remarks:
- Need to document that the DDF only works from firmware 2.4.45, or create multiple DDFs with a matchexpr on the firmware;
- Need to setup attribute reporting for _AC Power Divisor_ in line with reporting for _Active Power_, or the API plugin won't have the correct divisor.

ebaauw, 2024-11-30:
PR not yet merged.

NotSoScientific, 2024-11-30:
As per [my comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7948#issuecomment-2405379431) on 10-10, the suggested DDF definition in #7959  allowed the smartplug to be added via the REST api in Home Assistant. Currently waiting for the PR. Also, some elements seem to be missing compared to the ZHA implementation of this device, most notably the separate Voltage and Amperage listings. If a newer version of this is available, I would like to test it to see if it works. Where can I find it?

trustno1foxm, 2024-12-27:
Hello, where can I download that ddf file? thx!

mattreim, 2025-01-25:
For those who don't want to wait, I have created a simplified DDB. The basic requirement is the firmware: 2.4.45.

barsk, 2025-02-13:
I too has the problem with missing energy readingis in Inpelning. This is a hugely popular, good quality and cheap device, so I would indeed hope it gets merged soon.

barsk, 2025-02-14:
Yes sorry, I found the DDF file just after I wrote the comment above. I added it to my devices folder in 2.28.1, restarted and it was automatically applied to my Inspelning plug-ins. Great! As a side note as I am swedish, I can tell you that Inspelning means "Recording". :smiley:

One possible problem noted though. In Home Assistant when I monitor the Power entity I get negative readings (e.g. -230 W). 
 

[image]

Villhahallon, 2025-02-14:
@barsk 

I had a similar problem in HA.

Not sure how I solved it, but if you plan to add the sensor to the energy dashboard I had to make an helper integral that converted w to wh.

My kWh sensor also came in in corrupt for some reason...

ebaauw, 2025-02-14:
_Active Power_, the source of `state/power` is a signed integer, so in theory the device could report a negative value.  Could you please double-check the value in the deCONZ GUI, and the value reported by the API?

barsk, 2025-02-14:
In my case I have a fan on to read the power consumption, so it should not be negative? Or could it be that negative is **drawing** power and positive is **supplying** power (as in solar panels??? Maybe this is corect anyhow...

I get negative reading everywhere as you can se in my screenshots from Phoscon app and Home Assistand view. Not sure where else you want me too look. In particular, which "API"? f I open the Cluster Info for Elecrical Measurement I see there is a s16 variable (signed), but the raw data there seems to report a **positive** number, i.e 248 in my case here. 

[image]

I tried editing the DDF file and with both a sign conversion (-) in front of Math.round() and another try with Math.abs(), that should remove the sign. 
```
 {
          "name": "state/power",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0B04",
            "at": "0x050B",
            "eval": "if (Attr.val != -32768) { Item.val = -Math.round(Attr.val / R.item('state/power_divisor').val) }"
          },
```

But neither seems to stick. I tried several ways to **alter** the already loaded DDF:

- If I edit in the GUI and do an Hot Reload nothing changes with the readings in Phoscon App.
- If I try to load the inspelning_smart_plug.json file in the DDF Editor I get Failed to load error without any clue to what failed. It does give the same error regardless of what file there I try to load, even the original inspelning_smart_plug.json. Bug?
- If I edit the actual json file in /usr/share/deCONZ/devices/ikea/inspelning_smart_plug.json and restart nothing changes.
- I rename the inspelning_smart_plug.json to inspelning2_smart_plug.json and restart. Nothing changes.
- if I remove the plug-in nodes and re-pair them, nothing changes.

It seems the first version of the inspelning_smart_plug.json gets cached as the DDF editor refers to the inspelning_smart_plug.json file even if it is no longer there as I have renamed it. 

Are these DDF reload problems related to bug #6538 maybe?

So, I cannot find a way to test fixes to the DDF file unfortunately. Any ideas to that?

ebaauw, 2025-02-14:
> Or could it be that negative is drawing power and positive is supplying power (as in solar panels)?

That's usually the case.  I haven't seen any smart plugs that support this, thought.  As the GUI shows a positive value, I think it's safe to assume that the device reports a positive value as well.  To confirm, you would need to read the payload of the Zigbee messages in the deCONZ log, or use a Zigbee sniffer.

> In particular, which "API"? 

The deCONZ API, of course.  Do a GET of `/sensors/`_id_`/state` to see the value of `state/power`.  If both Phoscon and HA show a negative value, it's probably safe to assume that deCONZ reports this, but it might be through the API and/or through web socket notifications.

> I tried editing the DDF file and with both a sign conversion (-) in front of Math.round() and another try with Math.abs(), that should remove the sign.

That would address your symptom, but not the underlying problem.  And it might re-introduce the symptom, once the problem will have been solved, or automagically not longer occurs.

Note that `state/power` is updated at two places in the DDF: when _Active Power_ changes, **and** when _AC Power Divisor_ changes.  You would need to apply the workaround for both cases.  Also note that the API value will be updated only, when a new message is received from the device, not just on DDF reload.  Read the attributes in the deCONZ GUI to force a new message from the device.

> Are these DDF reload problems related to bug https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6538 maybe?

Probably.  I always restart deCONZ, to make sure the updated DDF is applied.  And I enable DDF/Javascript logging and check the deCONZ log, to make sure the new formulas are applied.

I would really like to understand where the negative value is coming from.  I've never seen this myself and I'm having a hard time understanding what type of bug in deCONZ might cause this.

Just to be sure: you did re-compile the REST API plugin from the latest commit (including PR 7979, for proper support of `state/power_divisor`), didn't you?

mattreim, 2025-02-14:
If the negative values ​​were due to the missing `state_power_divisor_item.json` with the fixed value of 10, it then worked. The latest firmware 2.4.45 should also be installed.

Home Assistant users can use this [DDF](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7959) or DDB. It only works properly up to 999W but as a transition it's ok.

[inspelning_smart_plug.zip](https://github.com/user-attachments/files/18800004/inspelning_smart_plug.zip)

The DDB can be read directly into Phoscon via the device integration.

barsk, 2025-02-14:
Ok, got the rest-api up and running with an api key.
I did not get the URl you mentioned working, but /sensors/<id>reported all of the data I think. Power is negative here.:
```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "a604115e1daec3d77d44b2ede663d59d",
    "lastannounced": "2025-02-14T13:28:46Z",
    "lastseen": "2025-02-14T15:25Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "INSPELNING Smart plug",
    "name": "Power 18",
    "productid": "E2206",
    "state": {
        "current": 424,
        "lastupdated": "2025-02-14T15:25:33.006",
        "power": -734,
        "voltage": 226
    },
    "swversion": "2.4.45",
    "type": "ZHAPower",
    "uniqueid": "94:ec:32:ff:fe:fe:dc:0e-01-0b04"
}
```

Also. I did NOT compile from source. I used apt-get in Raspberri Pi 3, bookworm 64 bit, with the stable release 2.28.1.
I then downloaded the DDF (json) file from github and copied it into the devices/ikea folder.

ebaauw, 2025-02-14:
It's not a bug - the DDF simply needs the changes from PR #7979 in the REST API plugin, as mentioned in the PR for this DDF, https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8025.

For compiling the REST API plugin yourself, see https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/BUILDING.md, but it's probably easier to wait for 2.29.2-beta.

barsk, 2025-02-15:
Ah, I missed that there was a required update for the rest api. My bad.
Thanks for digging in to this though. When is 2.29.2-beta supposed to be released?

sadpixel, 2025-02-18:
> Ah, I missed that there was a required update for the rest api. My bad. Thanks for digging in to this though. When is 2.29.2-beta supposed to be released?

When it’s done — and not a moment sooner 😉

slev1n, 2025-03-01:
Hi, I added the DDF via device integration within phoscon, but I'm unable to bind the plug. Plugs -> Add new plug -> on inspelning press little button for 5 sec. 
Im on latest HA iwht ConBee II (2.28.1)
Any suggestions?
Thanks in advance

mattreim, 2025-05-06:
@NotSoScientific This device is now supported and this issue can now be closed.

closed 2025-05-06 by NotSoScientific