dresden-elektronik/deconz-rest-plugin #7949: Cannot add NARON AS/4512703

olemr, 2024-09-28:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Seems like the DDF is not picked up and add fails.

### Steps to reproduce the behavior

Start adding the switch. GUI times out.

### Expected behavior

Add successful.

### Screenshots

[image]

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian
- Firmware version: 26400500
- deCONZ version 2.28.1
- Device: ConBee I
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? no

SwoopX, 2024-09-29:
Could you please try amending the respective entry for the model to use your manufacturername, so that is is all capital letters? There was a recent fix which should have restored case insensitivity on the manufacturername, but maybe it does not work as expected.

However, before you do that, please make sure to delete that device with a right click -> delete node and try to pair it again if it was previously paired and working .

olemr, 2024-09-30:
_Could you please try amending the respective entry for the model to use your manufacturername, so that is is all capital letters?_

I'm not sure which entry and where you mean here?

olemr, 2024-09-30:
That did the trick!
It would be helpful if the DDF/DEV log entries was a bit more talkative as to which DDF .json file it was picking up, and the result of the file parsing.
Do you want me to make a PR for the uppercase entries, or du you want to hunt down the case insensitivity yourself? (Beyond me ..) 
 
[image]

SwoopX, 2024-09-30:
Hi,

Many thanks for checking and sharing feedback. Actually, I'd tend to take this as a confirmation of a bug, that case insensitivity is not given.

Regardless, good that it works oit for you right now.

olemr, 2024-10-12:
@SwoopX did a fix make it to 2.29.0?
Didn't find it in the Changelog ...

SwoopX, 2024-10-12:
Thanks for following up. No, I'm afraid that has drowned a bit. I'll make a quick bug report later, so this matter is not disguised anymore and close/reference this issue.

In the meantime, you may update the version to your liking but if you do, do not forget to make the necessary amendments for the affected device.

SwoopX, 2024-10-13:
Closing in favor of #7965

closed 2024-10-13 by SwoopX