dresden-elektronik/deconz-rest-plugin #7955: Support new Hue light features in firmware v1.222.2

ebaauw, 2024-10-03:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Latest Hue light firmware v1.222.2 introduces four new effects: `underwater`, `cosmos`, `sunbeam`, and `enchant`.  These aren't yet visible in the Hue app, but they're listed in the v2 API.

### Steps to reproduce the behavior

Do a GET of `https://`_Hue bridge IP_`/clip/v2/resource/light`.

### Expected behavior

deCONZ to support the new features

### Screenshots

n/a

### Environment

- Host system: all
- Running method: all
- Firmware version: all
- deCONZ version (not Home assistant Addon version!): 2.28.1 and earlier
- Device: all
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

ebaauw, 2024-10-03:
Might also want to check the `signalling` feature, the differences between `effect` and `effect_v2`, adding a duration to the `timed_effects`.  And the new `taurus_7455` service.

bluemoehre, 2024-10-03:
Does it match the [post in the HUE blog](https://hueblog.com/2024/08/30/philips-hue-plans-four-new-effects-for-its-light-sources/)?

Effects: 
- **Cosmos** (_new_)
- **Enchant** (_new_)
- **Sunbeam** (_new_)
- **Underwater** (_new_)
- Candle
- Fireplace
- Glisten
- Sparkle
- Opal
- Prism

ebaauw, 2024-10-03:
I think so.  I see reports that the new light firmware might not yet be available everywhere, but in the Netherlands, the Hue bridge finds it.

bluemoehre, 2024-10-03:
In DE the update to v1.222.2 is available as well. (Unfortunately the Version Info Panel of the Hue App was empty - so I simply installed it to see the final version.)

closed 2024-10-07 by manup