dresden-elektronik/deconz-rest-plugin #7962: Deconz - Aqara Zigbee sensors are not refreshed for hours and keep a static state

MonkeyTrunky, 2024-10-10:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I am using Home Assistant OS on a Raspberry Pi 4 with deCONZ integration, deCONZ Addon and a RaspBee II Gateway. All worked fine for some years.

Suddenly all the Aqara temperature and humidty sensor didn't refresh anymore and kept a static state. They were not shown as unavailable but the state has not changed for hours.

The zigbee temperature sensors (integrated via deCONZ and raspbee II) are not refreshed. I don't have an explanation for this.

All connected temperature sensors stay at a static value at the same time. Shutting down Home Assistant and power cycle the hardware helps for a certain time but the problem is occurring again.

I have other devices in the network that are working fine. LIDL window sensor, LIDL switches, Philips Hue Dimmers, IKEA Styrbar Dimmers, IKEA blinds. --> The only issue is with the Aqara Multisensors and motion sensors.

### Steps to reproduce the behavior

1. I realized that the firmware of the RaspBee II was not up to date. I decided to migrate to a ConBee III and use the RaspBee II for another network and make the firmware update later.

2. I've flashed the latest firmware (26530900) to the ConBee III and uploaded the backup to it. I had to manually connect the sensors again. It worked for some minutes or hours and then I observed the same behavior -> static values.

### Expected behavior

Regularly refreshed values like usually at least after firmware update.

### Screenshots

[image]

[image]

[image]

### Environment

- Host system: (Raspberry Pi )
- Running method: Home Assistant OS, deCONZ Integration and Addon
- Firmware version: (26530900)
- deCONZ version: (2.28.1)
- Device: ConBee III /  RaspBee II (both with same issue)
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: Which? - SSD hard drive

### deCONZ Logs

00:14:32:517 ZCL attribute report 0x00158D000802B149 for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:14:42:665 ZCL attribute report 0x0C4314FFFE53A42D for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:15:01:951 ZCL attribute report 0x00158D0007E0B655 for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:15:02:268 ZCL attribute report 0x001788010C2AA81D for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
00:15:22:503 ZCL attribute report 0x0017880108F48347 for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
00:15:30:661 ZCL attribute report 0x84FD27FFFECF6AFA for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:15:43:448 ZCL attribute report 0x60A423FFFE053761 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:16:02:252 ZCL attribute report 0x00158D0008378574 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:02:332 ZCL attribute report 0x00158D0008378574 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:02:411 ZCL attribute report 0x00158D0008378574 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:05:148 ZCL attribute report 0x00158D000AF04E2E for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:05:276 ZCL attribute report 0x00158D000AF04E2E for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:05:323 ZCL attribute report 0x00158D000AF04E2E for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:31:350 ZCL attribute report 0x00158D000AF04630 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:31:396 ZCL attribute report 0x00158D000AF04630 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:31:411 ZCL attribute report 0x00158D000AF04630 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:49:908 ZCL attribute report 0x84FD27FFFED0DFCE for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:16:52:397 ZCL attribute report 0x00158D0007DF8A99 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:52:445 ZCL attribute report 0x00158D0007DF8A99 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:16:52:461 ZCL attribute report 0x00158D0007DF8A99 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
[... 62 lines cut ...]
00:32:24:237 ZCL attribute report 0xB4E3F9FFFE786606 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:32:38:929 ZCL attribute report 0x00158D000AF008E3 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:32:39:137 ZCL attribute report 0x00158D000AF008E3 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:32:57:895 ZCL attribute report 0x00158D000AF045FA for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:32:57:926 ZCL attribute report 0x00158D000AF045FA for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:32:57:958 ZCL attribute report 0x00158D000AF045FA for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:33:45:079 0x842E14FFFEE1BDD2 found group 0xFFF0
00:33:45:079 0x842E14FFFEE1BDD2 found group 0x0014
00:33:45:156 binding for cluster 0x0006 of 0x84FD27FFFECF6AFA exists (verified by reporting)
00:33:45:156 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x84FD27FFFECF6AFA (seems to be active)
00:33:56:864 ZCL attribute report 0x60A423FFFE050416 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:34:01:232 ZCL attribute report 0x842E14FFFEE1BDD2 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:34:03:882 ZCL attribute report 0x60A423FFFE0040D1 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:34:36:724 ZCL attribute report 0x00158D000AF04630 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:34:36:740 ZCL attribute report 0x00158D000AF04630 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:34:54:604 ZCL attribute report 0x0C4314FFFE53A42D for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
00:34:58:023 ZCL attribute report 0x001788010C2AA81D for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
00:35:01:143 ZCL attribute report 0x00158D0008378574 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
00:35:01:207 ZCL attribute report 0x00158D0008378574 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 

### Additional context

Network size: 59 devices
Gateway: 1
Brands: Aqara, LIDL, Philips Hue,  IKEA

MonkeyTrunky, 2024-10-16:
Since I needed a running Zigbee network again quickly, I switched to zigbee2mqtt and purchased a Sonoff ZB Dongle-E (EFR32MG21) and upgraded it to EmberZNet 7.4.4.

deCONZ and z2m are running in parallel at the moment, as I still have to add some devices to the Sonoff gateway that are still connected to the Conbee III. I have placed the two gateways at some distance from each other.

So far, everything has been very stable despite the same conditions as the Conbee III in terms of installation position, number of routers and distances. The framework conditions cannot be responsible for the static values that the Conbee III delivered.

[image]

TheNON75, 2024-11-01:
Hi, can you try downgrading Deconz to 2.22.00 or something similar but far away enough from your recent version? In my experiences older versions run aqara reporting properly.

bluemoehre, 2024-11-05:
The screenshots and description look similar to my current network issues, which are related to #7935. 

@MonkeyTrunky maybe you can check if your issues are caused by the DDF problem as well? (At least I am in the same situation as you; got the Sonoff Dongle-E today, struggled it to v7.4.4; now started moving nodes into the new network - what a pain)

MonkeyTrunky, 2024-11-09:
> Hi, can you try downgrading Deconz to 2.22.00 or something similar but far away enough from your recent version? In my experiences older versions run aqara reporting properly.

Hi,

Unfortunately I cannot do as suggested. I struggled for weeks finding a solution and asked the deCONZ support several times for help. I haven't received an answer at all.

After this experience, I switched to zigbee2mqtt with the Sonoff ZB Dongle-E. Everything works fine now. As the heating season started and I was dependent on the room temperatures, I didn't have time to wait any longer for a response from Dresden Elektronik.

And I refuse to downgrade to older firmware. In my opinion, this is not the right approach. The aim should be to keep the current firmware compatible. If Dresden Elektronik has this sensor in its own compatibility list, then it should also work. Otherwise it should no longer be listed there.

MonkeyTrunky, 2024-11-09:
> The screenshots and description look similar to my current network issues, which are related to #7935.
>
> @MonkeyTrunky maybe you can check if your issues are caused by the DDF problem as well? (At [...]

@bluemoehre Thank you for sharing your report.  I hope Dresden Elektronik is going to fix it. It should be in their interest. And I agree with you: I also think this should be top priority. In my opinion this is a bug. Maybe older firmware can fix it, but this is not the way I want to go. Apart from that, I'm pretty excited about how powerful zibee2mqtt is.

TheNON75, 2024-11-09:
@MonkeyTrunky 
Hi, I did not say to downgrade firmware. I was talking about software version, the two things are different kind of things. Firmware is on the stick, software runs on your computer. With docker it is just 1-2 minutes of total effort to downgrade.
Anyhow you have the solution you were looking for and it is great as you won't suffer.

MonkeyTrunky, 2024-11-12:
Hi,
Aha, then I misunderstood. That would of course be an option. You're right, I now have a solution that I'm happy with. Thank you for your advice and the patience.

EmlJnke, 2024-11-24:
> Hi, can you try downgrading Deconz to 2.22.00 or something similar but far away enough from your recent version? In my experiences older versions run aqara reporting properly.

Hi @TheNON75 
I also ran into this bug, but downgrading the firmware to the version you suggested solved the problem (one of my Aquara sensors not connecting). Thank you :)

bruesche, 2024-12-13:
Hi there,

i have the same problem. The Aqara sensors are no more operational. Stop and start of Deconz let them work again for a short time but give only the value not the time of last refresh. I want to downgrade the Deconz Software to 2.28.1 but get dependencies errors and then i can only perform a recover from a backup. What can i do?

or is there an other workaround?

joshefin, 2024-12-13:
I think this issue should have very high priority. Aqara contact sensors are widely used in smart home setups.

FrankyB77, 2024-12-16:
Hi, i have the same issue. I hope there will be a fix soon!

MSchmitt67, 2024-12-20:
Hallo,
i have the same problem with Aqara Door Windowsensors

bruesche, 2024-12-20:
Hi, 

can't go back to lower version because of dependency errors. Don't know why. For me the door and window sensor for heating solution. So I write a small script to stop and start the deconz process every 30 min. It's ok for me for the moment. The Door sensors work after restart the process and the process lives long enough so the heating solution work. The temperature sensors should work too. If the sensors no more sent the actuall value every 30 min the process restarts. 

So I can now wait for solutions in deconz software.

prebbz, 2024-12-22:
I have the same issue with my Aqara sensors - I can force them to reconnect, but then they will just stop working (even though they don't report as being unreachable).

prebbz, 2024-12-25:
@bruesche How is it acceptable that deconz should need to be restarted every 30 mins? This really is quite a serious bug, I have forced docker to use 2.25.3 (only hope that the SQLite database hasn't changed between these versions!!!) and will see if it improves.

prebbz, 2024-12-25:
This will be a last resort - but I am almost tending to want to want to try this:
https://www.creatingsmarthome.com/index.php/2024/02/25/moving-from-deconz-to-zigbee2mqtt/
Things like no OTA updates for my devices, and not being able to update the deconz stick fw in docker are actually quite limiting

Mimiix, 2024-12-25:
> This will be a last resort - but I am almost tending to want to want to try this: https://www.creatingsmarthome.com/index.php/2024/02/25/moving-from-deconz-to-zigbee2mqtt/ Things like no OTA updates [...]

You can do both ;)

MonkeyTrunky, 2024-12-25:
I moved all devices to zigbee2mqtt. Everything works smoothly now. Pairing is a lot easier, too. And z2m supports many devices, also the ones that were not compatible with DeConz.

bruesche, 2024-12-25:
> @bruesche How is it acceptable that deconz should need to be restarted every 30 mins? This really is quite a serious bug, I have forced docker to use 2.25.3 (only hope that the SQLite database hasn't changed between these versions!!!) and will see if it improves.

@prebbz it is Not acceptable, but this is for me the best way for the Moment. Have an Aqara M3 Hub here, so I think about swapping the Sensors to the Hub. 
But Need more Time to check this way

prebbz, 2024-12-26:
@bruesche I could also buy a Aqara hub, but the whole point/the beauty of deconz is to reduce the number of IoT hubs one needs to maintain to just one.
@MonkeyTrunky  it's just such a project to move everything over and deconz has been on the whole very good for many years.

Update: forcing v2.25.3, my Aqara sensors are updating again.

MSchmitt67, 2024-12-26:
i am wondering if there will be a fix in the next update. No reaction from anyone :(

MonkeyTrunky, 2024-12-26:
My experience with deconz is anything but flawless. I can only recommend zigbee2mqtt. With z2m, only one hub (coordinator) can be used for all devices, too.

Supported adapters:

[https://www.zigbee2mqtt.io/guide/adapters/]()

In addition, z2m's documentation is very good, which is more than I can say for deconz. And if I don't get any response from deconz support, I think that's a shame.

That's just my experience. If you find an acceptable solution with deconz, that's good too.

FrankyB77, 2025-01-05:
I updated NodeJS from version 18.x to 20.18.1 on my Raspberry, which also runs Phoscon 2.28.1. I also updated the Raspberry Pi OS. Since then, the Aqara Weather sensors have been updated regularly again. However, I don't know if it has anything to do with that, but for now it's working.

bruesche, 2025-01-20:
Hi there,

Is there anything new about this error? When it will be fixed or if it will be fixed at all?

Thx for Info

Bruesche

MonkeyTrunky, 2025-02-11:
@Dresden-Elektronik any reaction would be welcome.

MSchmitt67, 2025-02-11:
i am still on 2.25.3 and no problems at all

pdeliot, 2025-02-11:
Any investigation?

bluemoehre, 2025-02-11:
@MonkeyTrunky 
> After this experience, I switched to zigbee2mqtt with the Sonoff ZB Dongle-E. Everything works fine now. As the heating season started and I was dependent on the room temperatures, I didn't have time to wait any longer for a response from Dresden Elektronik.

Just to let you know, your comment motivated me in moving asap. I finished a few weeks ago and migrated hundreds of devices to Zigbee2MQTT. It was a huge effort and took multiple weeks to re-pair all devices. But I got rid of all these stability problems and the network had no hick-ups so far - also it feels way more performant. Sure, there are some different issues in Z2M and a lot of things to learn upfront. But every time I faced an issue it was solved within a few days. I absolutely love Z2M was immediately upgrading 30 devices of different brands to their latest firmware.

MonkeyTrunky, 2025-02-11:
Thank you for your feedback :) I am still happy with z2m and have never regeretted it. What is unbeatable about z2m is the extensive documentation and as you mentioned the performance. MQTT opens new possibilities as for example mqtt in/out node in node red. It is pretty awesome :)

prebbz, 2025-02-15:
Like several others in this thread, I must say I am extremely disappointed in the lack of engagement from the deCONZ community on this issue and I have started the process of phasing out deconz. Firstly by buying a new compatible adapter (SonOff ZB Dongle-E also), following the (extremely well documented) setup guide ..  I will run the two networks in parallel and gradually move devices over to the Zigbee2Mqtt. 

I fully realize is opensource, but the "nonchalance" over this makes it worth the effort to try something else

prebbz, 2025-02-16:
To continue my own comment - I can't believe I didn't try this earlier!!!! Devices that either never worked with deconz (e.g my IKEA blinds) or were an absolute nightmare to pair (Schnieder PIR dimmer!) ... a breeze to add to Zigbee2MQTT. 
I love how each device has the appropriate picture rather than a generic ugly bulb symbol.. I love that the properties/capabilities of the devices are easily viewable ... this is still perhaps the honeymoon period, and my Home Assistant is currently in ruins.. but I feel this is a giant leap forward and I strongly advise others to at least look at Zigbee2MQTT

MonkeyTrunky, 2025-02-16:
I can tell you from my experience. It is worth the effort. I am very satisfied. The device compatibility is very good. I was also able to pair devices that were not possible with deCONZ. The network map is also better than with deCONZ. And via MQTT, very short response times are possible for automations or Node Red. I have built a dimming function with a rotary switch. It was a bit too slow with deCONZ.

MonkeyTrunky, 2025-02-17:
> Hallo,
> i have the same problem with Aqara Door Windowsensors
> 
> 2.28.1 = problem
> 2.27.6 = problem
> 2.25.3 = works since 6 days

MonkeyTrunky, 2025-02-17:
> i am still on 2.25.3 and no problems at all

bluemoehre, 2025-02-18:
Just to let you guys know ... it's not network size related.
I only have 2-3 sensor devices left on the deCONZ integration (cuz they are hard to reach). With the current version, I experienced hours of massive dropouts again that require a manual restart of the service.
The same sensor type works flawlessly in Z2M via Sonoff E.

prebbz, 2025-02-19:
> To continue my own comment - I can't believe I didn't try this earlier!!!! Devices that either never worked with deconz (e.g my IKEA blinds) or were an absolute nightmare to pair (Schnieder PIR [...]

There are some things I do miss from Phoscon, and thought I would share them as a caveat for those that are considering jumping ship: Switches. These were a breeze to setup on deconz, and binding, button assignment etc was nicely handled by the Phoscon app. Zigbee2Mqtt constantly loses contact with my philips hue dimmer switch (both of them) and its not a routing problem as there are plenty of router devices for it to connect. 
Binding buttons seems more cryptic with Zigbee2mqtt as well.. 

The other issue I have is some devices dropping off, in this case mainly the switches..

bluemoehre, 2025-02-20:
@prebbz are you talking about [this](https://github.com/Koenkk/zigbee2mqtt/issues/26227)?

MSchmitt67, 2025-02-23:
> > i am still on 2.25.3 and no problems at all

me too.

I hope their will be a fix in the next version.

MonkeyTrunky, 2025-03-11:
> There are some things I do miss from Phoscon, and thought I would share them as a caveat for those that are considering jumping ship: Switches. These were a breeze to setup on deconz, and binding, [...]

Interesting. I have a Philips Dial Switch connected with no issues at all.

If you haven't already checked this:
Zigbee and WiFi can cause interference, which then leads to poor Zigbee and WiFi connectivity. If the router changes the channel, this can lead to sporadic problems, which then disappear again without your intervention as soon as a non-critical channel is active. 

I have switched off Auto Channel on my router and set a self-selected channel that is outside the interference range of the set Zigbee channel. This has greatly improved the signal quality for both Zigbee and WiFi. You can see which channel works best for you in the picture.

[image]

GatuRatz, 2025-03-14:
Hi,
I have the same problem with software version 2.29.2. Is there any fix planned? 
Or is there a way to fix it by myself (without downgrading)?
I do not want to buy a new zigbee-dongle or new sensors (I have plenty of them).

Br
GatuRatz

schoen2, 2025-03-14:
Hi @GatuRatz, I had actually the same issue.

for me, with version 2.29.2 I ran in the same issues last weeks. Today, I tested this with several older versions (like 2.29.1, 2.29.0, ..).
I downgrade back to version 2.25.3 and everything works fine.

I hope, there will be some progress regarding this issue.

GatuRatz, 2025-03-14:
Did you try to use some DDF from https://deconz-community.github.io/ddf-tools/#/?

EmlJnke, 2025-03-16:
2.25.3 has the same bug for me, the devices stop updating after a really short period of time. Hope there will be a fix soon!

GatuRatz, 2025-03-17:
For me 2.25.3 works fine at the moment. 
However I do not want to use old software versions, they might have other issues. I have the impression that this will not be fixed and I also had issues with a blind from Ikea (which I could fix by myself).
There is also some hardware around here which I could never get to work at all with deconz.
I bought a stick for zigbee2mqtt which will arrive today and I will migrate all the heating system related sensors and actors to this asap. 
The other stuff will follow with less urgency (once the heating stuff works fine).

schoen2, 2025-03-17:
> Did you try to use some DDF from https://deconz-community.github.io/ddf-tools/#/?

Not yet, but I will do it, this week. The alternative for me is also to buy a zigbee2mqtt stick. It´s really frustrating to use old software in times of RenovateBot/Mend.

GatuRatz, 2025-03-17:
I did not have sucess with the DDFs. 
And I also noticed issues on some sensors from SONOFF, so meanwhile I believe that it is not limited to Aqara but rather some general problem which affects all of these sensors.

nikolaykasyanov, 2025-03-29:
I couldn't roll back because otherwise my Ikea Inspelning plug wouldn't work, so I tried DDFs from the site mentioned above, and now my Aqara sensors are working fine.

GatuRatz, 2025-03-29:
For me they worked during some hours with the new DDF... Which version of deCONZ are you using now and during how much time is it working for now? 

- Anyway, I changed all my devices to zigbee2mqtt and everything is stable now... As part of them are used by my heating system (also the actors) I see no option of doing experiments during a longer time...

schoen2, 2025-03-31:
@nikolaykasyanov, can you confirm, that your sensors working for more than a few hours? I will try the DDFs this week and come back here.

nikolaykasyanov, 2025-04-01:
@schoen2 I can confirm it:
[image]

schoen2, 2025-04-01:
Ok, thanks. This confirms also, that I´m to stupid to read the DDF docs :D

I placed the DDF as a .ddb-file in the /opt/deCONZ/devices directory inside my Docker-Container. I use Deconz as a Docker deployment and mount /opt/deCONZ as local volume.

Euestros, 2025-04-11:
Sorry to hijack this thread, but @schoen2 , @nikolaykasyanov  could you possibly share how you managed to get this working?
I've read through the DDF documentation, but I'm still too stupid to figure it out properly.

I'm currently running gateway version 2.28.1, and placed the DDB file "Temperature and humidity sensor (WSDCGQ11LM)" downloaded from https://deconz-community.github.io/ddf-tools/ into the path ~/.local/share/dresden-elektronik/deCONZ/devices on my Raspberry Pi with deCONZ.

After rebooting, the sensor starts updating correctly - but only for about an hour. After that, the updates stop entirely until I reboot again.

bruesche, 2025-04-17:
Hi there,
The new Beta 2.29.4 has fixed this issue:

_**Fix hot reload and events not emitted after joining a device https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8178**_

Is this a Solution for the aqara sensors too?

Thx for help

manup, 2025-04-17:
It's worth a try. The v2.29.4-beta together with firmware verion 0x26550900 (ConBee 3) should runt smoother due the various bug fixes.

schoen2, 2025-04-17:
I habe updated my instance to 2.29.4, 2h ago. Until now, everything works. But i will give feedback again this evening.

schoen2, 2025-04-18:
Seems to work for me. My setup: Kubernetes deployment with Docker-image 2.29.4, ConBee 2 with firmware 0x26780700 and DDF/DDB Aqara lumi.weather (WSDCGQ11LM).

EmlJnke, 2025-04-21:
@schoen2 Do the sensors still work for you? I changed my system to Zigbee2MQTT but I really like the Deconz features (like the temperature history in the eve home) app so I want to go back eventually. However, I only want to do that once the bugs are fixed on the deconz side!

schoen2, 2025-04-21:
Hi @EmlJnke, yep, with 2.29.4 everything works for me - sensors (aqara), plugs and switches.

MSchmitt67, 2025-04-28:
i updated 4 days ago to 2.29.5 and no problems at all.

manup, 2025-04-28:
Alright thanks for the feedback 👍  I think we can close this issue for now.

closed 2025-04-28 by manup