dresden-elektronik/deconz-rest-plugin #7968: Phili[s Hue Smart Plug

pimw1, 2024-10-14:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Philips Hue Smart Plug

### Manufacturer

Philips

### Model identifier

LOM001

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]
[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]
[image]

### Level Control
[image]
[image]
[image]

### Color Control
NA

### Thermostat
NA

### Simple Metering
NA

### Electrical Measurement
NA

### Any other cluster of relevance/interest
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2024-10-14:
What's the issue with this device? As it seems to be working?

pimw1, 2024-10-14:
There is no DDF while there should be one (see the discord "Advice"  channel and Erik's response)

ebaauw, 2024-10-14:
This is indeed a different model from the existing DDF.  What’s the _ Image Type_ in the _OTA Upgrade_ panel?  This seems to be an ancient model, or at least ancient firmware.  Is it an EU plug?  Did you try and upgrade it’s firmware?

pimw1, 2024-10-14:
The image type is 0x0000. It's just a regular EU version of the plug, bought in Dutch well known retail store around 2020 i think. I've never tried to update the firmware, since that is quite difficult to do if you don't have the Hue Bridge.
[image]

ebaauw, 2024-10-14:
Make sure to select the node, and press _Query_ to populate the _Image Type_.  deCONZ is perfectly able to update the firmware, the challenge is finding out where to download the file from.

pimw1, 2024-10-14:
Pressing _Query_ does not change the image type, it's like nothing is happening

ebaauw, 2024-10-14:
You need to check the line in the table; the attributes above are only populated when a firmware file has been selected to upgrade the device.

pimw1, 2024-10-14:
[image]
The line is selected in the table, which is indeed the plug of interest. _Query_ does not change anything. I've no firmware file selected , is that also a requirement before the image type can be read?

ebaauw, 2024-10-14:
The table shows the existing firmware as reported by the device.  The fields above show the selected file for the selected node.

Yes, it's confusing, and we changed the column headers to better fit the available space.  _Manufacturer Id_ and _Vendor_ are the _Manufacturer Code_ (0x100B for Hue); _Image Type_ and _Image_ are the _Image Type_ (0x0115) and _File Version_ and _Version_ are the _File Version_ (0x0100040A).   The Hue app reports the _Hardware Platform Version_ as 100b-115, which is made up from the _Manufacturer Code_ and _Image Type_.  Note that the _OTAU_ cluster reports the _Manufacturer Code_ and _File Version_, but not the _Image Type_ (it shows the default 0xFFFF).  The _Manufacturer Code_ is also reported through the _Node Descriptor_.

pimw1, 2024-10-14:
Aah ok. Based on this info, i think i should use this image (0115 -> firmware 1.93.6). I'll try that and report back.
[image]

pimw1, 2024-10-14:
I've places the .zigbee file in the otau folder , selected it in the gui and started the update proces for the selected zigbee plug. The Progress column keeps showing "Idle". Is there any way to force this?

pimw1, 2024-10-14:
I just realize that the smart plug has also bluetooth. I'll use the Hue app to update the firmware and report back.

ebaauw, 2024-10-14:
It should find file version 0x01001402.  My guess would be that the version in the _Basic_ cluster will be 1.222.2.

ebaauw, 2024-10-14:
We could try and add it to the DDF for the LOM007.  Despite different firmwares, the devices probably behave the same.  Just to be sure, could check the clusters after the upgrade: it should now expose FC03 and FC04 and no longer FC02.  Make sure to refresh the endpoints and simple descriptors, so the GUI reflects the new list.

pimw1, 2024-10-14:
I've refreshed the endpoint and simple descriptors twice, restarted deconz and repeated the refresh of the endpoint and simple descriptors again. Restarted deconz again and repeated yet another time.

It still displays FC02
[image]
[image]

pimw1, 2024-10-14:
Can i use this file 
[image]
...in the Phoscon ui to try it out?

[image]

ebaauw, 2024-10-14:
Not sure, it's part of a bigger PR.
If it doesn't work, just to edit `/usr/share/deCONZ/devices/philips/lom007_smart_plug.json`.

pimw1, 2024-10-14:
Both unsuccesful
- Phoscon: can only upload .ddf, not .json
- Copied the json to (rights set 775)
[image] 
following error occurs when trying to open it in Deconz:
[image]

pimw1, 2024-10-14:
...or maybe not. When reapplying the old settings (-v /etc/localtime:/etc/localtime:ro   and removing the environment variable), the DDF is still there.

It is pointing to the .json i've got from you
[image]

So, somehow, while the initial load of the json fails, when restarting the docker container, it starts to pick it up.
Weird behaviour. The plug is working fine. Also my other 2 plugs (which are still with the very old firmware) are working as expected with your json file.

pimw1, 2024-10-15:
All in all, i think the conclusion is the following:

- the new json works as expected. Thanks!
- there is a unrelated issue with loading jsons in the deconz gui, which can be bypassed by restarting the deconz docker container

ebaauw, 2024-10-15:
Please open a new bug report issue for the second point.  I don't use Docker and cannot help here.

pimw1, 2024-10-15:
Done: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7971