dresden-elektronik/deconz-rest-plugin #7969: Philips Hue Lightguide E27 Edison ST72

pimw1, 2024-10-14:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Philips Hue Lightguide E27 Edison ST72

### Manufacturer

Signify Netherlands B.V.

### Model identifier

LCV001

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]
[image]
[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]
[image]
[image]

### Level Control
[image]
[image]
[image]

### Color Control
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Thermostat
NA

### Simple Metering
NA

### Electrical Measurement
NA

### Any other cluster of relevance/interest
Identify
[image]

Groups
[image]
[image]

Scenes
[image]
[image]
[image]
[image]

Touchlink
[image]
[image]

Hue Effects
[image]

Hue Entertainment
[image]

Hue Unknown
[image]

OTAU
[image]
[image]

Green Power
[image]

pimw1, 2024-10-14:
No DDF available yet

ebaauw, 2024-10-14:
Also for this one: what’s the _Image Type_?  And you might want to upgrade it’s firmware; I expect it will support more special effects with newer firmware.

pimw1, 2024-10-14:
Image type is 0x0000
[image]

I'll update the firmware in a bit via bluetooth and send the new printscreens

ebaauw, 2024-10-14:
0x0114 is a familiar one.  I'll add this model to `light_zb3_C.json`.

pimw1, 2024-10-14:
Great, thanks! This is the new info after updating to firmware 1.122.2

Basic
[image]
[image]
[image]
[image]

Hue Effects
[image]

pimw1, 2024-10-15:
I've tested the light_zb3_C.json with the Lightguide, most features seems to be working, but one is not working: the option to set brightness/color of the light while off.

[image]

`pydeconz.errors.pydeconzException: {'address': '/lights/18/state/bri', 'description': 'parameter, bri, is not modifiable. Device is set to off.', 'type': 201}`

ebaauw, 2024-10-15:
Did you set `config/bri/execute_if_off` and `config/color/execute_if_off`?

pimw1, 2024-10-15:
Whoops, i did not set that (i totally forgot about it). It works as expected now.

closed 2024-11-13 by github-actions[bot]