dresden-elektronik/deconz-rest-plugin #7971: Deconz: failed to open DDF

pimw1, 2024-10-15:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

When pressing control-e to open the DDF editor, followed by control-o, selecting the appropiate json.file, the following message appears: failed to open xxxxx.json

The DDF will be loaded by either choosing "Hot reload"  or by restarting the deconz docker container.

### Steps to reproduce the behavior

1. select a device
2. control-e
3. control-o
4. select the appropiate .json file
5. observe the error
6. control-r to hot reload the DDF

### Expected behavior

1. select a device
2. control-e
3. control-o
4. select the appropiate .json file
5. not giving an error but either "auto-hot reload" for me or display an info message that i should hot-reload

### Environment

- Host system: Proxmox NAS, running a Ubuntu VM
- Running method: deconz-docker container
- Firmware version: 26780700
- deCONZ version: 2.29.0
- Device: ConBee II (connected via USB 2)
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? 2x HDD via USB 3

Subnum12, 2024-11-11:
👍   same issue
also if you save a custom DDF, you cannot open it anymore. No error in log btw.

[image]

also [documentation ](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet#precedence-of-ddf-files) seems incorrect.
>directory of the user running deCONZ (e.g. `/home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices`) should be `/home/<DECONZUSER>/.local/share/deCONZ/devices`

pimw1, 2024-12-03:
Given the lack of response, i think this has no priority for Dresden Elektronic. Therefore, i am closing this issue.

closed 2024-12-03 by pimw1

kimchiii0411, 2025-01-18:
I'm having the same issue. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/8048

Smanar, 2026-02-28:
Ok, have the same issue
Not possible to load DDF, I need to restart deconz, and ofc deconz is hanging https://forum.phoscon.de/t/cannot-stop-deconz-via-systemctl/6723/12

Version 2.32.4 beta