dresden-elektronik/deconz-rest-plugin #7975: Lights groups disappear from phoscon (hassio)

PaoloVittori, 2024-10-16:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I've added 2 lights to a group under Phoscon.
I've added a wireless switch (BTicino K4003DCW)  to the network (not to the group).
I searched for the group id in the Phoscon app under help->API information
With deconz->Bind dropbox I linked the On/Off and Level clusters of the wireless switch to the id (in Hex format) of the group.
Everything works properly, I can turn on/off and adjust the intensity of the light with the button.
After a system restart, the group disappear from Phoscon app and from API information file too but light is still correctly controlled by the wireless switch.
Tried several times (with differents light and switces) with the same result. What happens?

### Steps to reproduce the behavior

1. Create a light group under Phocon.
2. Join a K4003DCW double rocker switch to the network.
3. Under Deconz bind the K4003DCW cluster to the group address (get from API information file).
4. Here I added some more property to the light group (Area, Label) under hassio (settings->Devices), but I'm sure this is not relevant.
5. Chek tha everithig is OK, the light are controlled by the switch.
6. Restart hassio.
7. Light Gruop disappear from Phoscon and from API information file but the lights are still controlled by the switch.

### Expected behavior

Light groups present in the Phoscon app

### Environment

- Host system: x64 platform
- Running method: (Home Assistent deCONZ Add-on)
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.28.01)
- Device: (ConBee II )
- Do you use an USB extension cable: (yes)
- Is there any other USB or serial devices connected to the host system? No

Smanar, 2024-10-16:
Hello, I m just seen they have removed the old webapp to make bind, but it's not possible using Phoscon ? (Instead of deconz direclty)

PaoloVittori, 2024-10-16:
With Phoscon I have never managed to bind on the K4003DCW switch. 
The problem of the group disappearing on Phoscon also remains on groups that have not bind any control device.
Additional note, if I log out  (Phoscon log out) and log again by entering the password, I see all the groups, some of which however disappear if I change the page and then return back.
What's even stranger is that if I use a REST-API client add-on to access the API, with the GET request 
'http://<ip_add>:<port>/api/XXXXXXXXX/groups, I get the list of all the groups created even if I don't see them in the Phoscon app

Smanar, 2024-10-17:
>With Phoscon I have never managed to bind on the K4003DCW switch.

Honnestly IDK how to do without the old webapp.

You are sure the problem is not just from Phoscon ? Your third app (HASSIO) can use group too ?

PaoloVittori, 2024-10-18:
Yes, in hassio I see the groups correctly and I can use them. 
The wireless switches binded to the groups (clusters on/off and level control binded in deconz) also work correctly. 
The problem is just with Phoscon.
The only way I can get the group id is via REST-API.

SwoopX, 2024-10-20:
Hi,

As this is not a bug report as described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5113, I am closing this issue.

Please feel free to open a topic in the forums, as this would be the appropriate place to highlight/discuss this matter. Thanks for understanding!

closed 2024-10-20 by SwoopX