dresden-elektronik/deconz-rest-plugin #7983: b-parasite

elratt0r, 2024-10-19:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

b-parasite

### Manufacturer

b-parasite

### Model identifier

b-parasite

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### Temperature
[image]

### Relative humidity
[image]

### Soil moisture
[image]

### Illuminance
[image]

### Electrical Measurement

### Any other cluster of relevance/interest

elratt0r, 2024-10-19:
DDF in #7972

wvuyk, 2024-10-20:
If you look at the device in the map, it is already active as "DDF" is mentioned? You should be good to go with it?

elratt0r, 2024-10-20:
That is the DDF from the referenced PR. I had it running for a few days to be sure it works sufficient.

SwoopX, 2024-10-20:
@elratt0r Thanks! 👍

elratt0r, 2024-11-11:
Issue solved in #7972 , waiting for merge

elratt0r, 2025-04-11:
Approved PR still waiting for merge, please reopen.

closed 2025-04-14 by manup