dresden-elektronik/deconz-rest-plugin #7990: Moes AM43 Blind Drive Motor _TZE200_icka1clh

tobiasr19, 2024-10-21:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

ZigBee AM43 Blind Drive Motor

### Manufacturer

_TZE200_icka1clh

### Model identifier

TS0601

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

Maybe similar to [issues/4521](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4521)

unix9t9, 2024-11-06:
I also have bought These Moes am43 Blinds Drive because the blakadder repository said it should work with deCONZ. Mine have another Manufakturer. It is: _TZE200_zah67ekd. The rest seems to bei the same like tobiasr19 blinds driver. My motors are recognized as a battery sensor. Repairing without deleting the node or reading the Basic Cluster or reading the node and simple discriptors hast no effect. Maybe someone can make a ddf for it?
[image]
[image]
[image]
[image]
[image]
[image]

tobiasr19, 2024-11-06:
It works out of the box with zigbee2mqtt. Change from deconz to zigbee2mqtt was my solution.

Smanar, 2024-11-06:
`    {"_TZE200_zah67ekd", "TS0601", "MoesHouse / Livolo", "Tuya_COVD AM43-0.45-40"},`

This one is working natively on deconz, I don't remember the procedure to include it, but you have it on a post on the forum less than 1 month ago.

unix9t9, 2024-11-06:
> ` {"_TZE200_zah67ekd", "TS0601", "MoesHouse / Livolo", "Tuya_COVD AM43-0.45-40"},`
> 
> This one is working natively on deconz, I don't remember the procedure to include it but you can try that to start #7958
> 
> or
> 
> > You need to incluse two Times first in phoscon, and the second Time, in deconz, you need to enable joint and press right click on the module and Made the actions 1, 7 and 8 (in the joint Time)

Thank you. I will try IT the next days... What du you mean with making the actions 1 7 and 8? Alt + Cntrl + 1 or 7 or 8?

unix9t9, 2024-11-12:
I tried to pair the blind Drivers as you descriped it. But it hast no effect. The node stays a battery sensor. There is No new entry in the API created. I tried it several times. I also paired it twice without deleting the node. But nothing happened.

Smanar, 2024-11-12:
Honneslty I don't remember the procedure, you have tried with deleting the battery entry ?
You don't have access to the GUI ? It's easier to force the inclusion with it.

unix9t9, 2024-11-12:
I have Access to the GUI.I deleted the node. Repaired IT twice an tried to do the shortcut thing you discribed. But no Chance. Maybe i try IT again tomorrow

unix9t9, 2024-11-13:
I got it. The Solution was do restart the Gateway after pairing and set it Info permit Join again. Then read the discriptors.

unix9t9, 2024-11-13:
Is there a possibility so read the battery Status of the Blinds driver?

Smanar, 2024-11-14:
The ZHABattery device is not created this time ?

unix9t9, 2024-11-14:
It ist created, but it sends no Status. It ist Always NULL

closed 2024-12-13 by github-actions[bot]