dresden-elektronik/deconz-rest-plugin #7991: [Bug] Parameters missing for lumi.sensor_motion.aq2 Aqara RTCGQ11LM : Temperature, Voltage, Power Outage

Ltek, 2024-10-21:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

For lumi.sensor_motion.aq2 Aqara RTCGQ11LM, Deconz is not exposing all the parameters available, Z2M exposes them. Missing several but Temperature is most important missing parameter.

Looks Temperature was possibly removed and supposed to be put back in a long while ago, looks like it was not?... https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6203

IMO Parameters should always be exposed to a home automation system if they exist. The admin can determine what/when to use the parameters. Everyone does HA differently, that is the beauty of HA, please do not limit our usage of the devices. thank you

### Steps to reproduce the behavior

Install one of these devices. The screenshot shown, the bottom device is this exact same device physical device that I joined to Z2m (shows 6 built-in parameters), then joined to Deconz (shows only 3 parameters)

### Expected behavior

expose all available parameters to home automation systems

### Screenshots

[image]

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / deconz-docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

Mimiix, 2024-10-21:
Please provide all details about the environment./

@SwoopX I remember that this was on purpose as it is not really reliable. can oyu elaborate?

SwoopX, 2024-10-21:
First of all, it is generally not a good idea to report a bug on software A by what software B is making out of A. I'm not sure where the screenshot is coming from and what is is supposed to tell us, but it doesn't reflect what the deconz REST API is exposing.

That being said, I checked what the device is actually exposing as manup and I just recently used it for some extensive testing. Everything what is expected to be there was found, and that is in sum even more than on the screenshot. It is perfectly in line with the device documentation [here](https://dresden-elektronik.github.io/deconz-rest-doc/devices/xiaomi/xiaomi_rtcgqq11lm_presence_sensor/) This means with one exception due to the recent changes and consequently a not yet updated documentation: the referred to temperature

The device does not support any temperature and it has been removed (again) with the last release https://github.com/dresden-elektronik/deconz-rest-plugin/releases. On a side note, Aqara has also no temperature in their app/gateway.

To sum up, I'm afraid this is not a bug report as described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5113, and therefore I am closing this issue.

Please feel free to open a topic in the forums if you feel this is necessary. Thanks for understanding!

closed 2024-10-21 by SwoopX