dresden-elektronik/deconz-rest-plugin #7996: Philips Hue Play Lightbar (LCT024)

bluemoehre, 2024-10-26:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Hue Play Lightbar

### Manufacturer

Philips

### Model identifier

LCT024

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]
[image]

### Color Control

[image]
[image]
[image]
[image]
[image]
[image]

### HUE Entertainment

[image]

ebaauw, 2024-10-26:
You didn’t post the _Basic_ cluster, but given that it doesn’t carry the FC03 cluster, this light also has old firmware.  Also here: what’s the _Image Type_?

bluemoehre, 2024-10-26:
> You didn’t post the Basic cluster […]

Copy-paste error … I updated the description.

> This light also has old firmware

Have you seen a newer image somewhere? Since this device is without BT I would need an OTA file. The current version is the latest I find.

ebaauw, 2024-10-27:
Try https://firmware.meethue.com/v1/checkUpdate?version=16783620&deviceTypeId=100b-11e

bluemoehre, 2024-10-27:
Looks to me the image types do not match. Anyhow I added the file, but the device does not update.

ebaauw, 2024-10-27:
My bad.  It's `0x010e`, not `0x011e`.  And that _Image Type_ isn't listed on firmware.meethue.com (confirming the assumption that the site is used for BLE updates only).  There is a version ``0x01002600` listed on https://github.com/Koenkk/zigbee-OTA:
https://github.com/Koenkk/zigbee-OTA/blob/c1c4488759b9ef68f7d0bc8c649ca8f306f4478b/index.json#L187-195

bluemoehre, 2024-10-27:
> There is a version 0x01002600

Many thanks - gonna update screenshots when update is done

> And that Image Type isn't listed on firmware.meethue.com (confirming the assumption that that site is used for BLE updates only).

Is it like that? Always thought its related to _when_ a device was manufactured - all the "original" Philips are on the old server and all the new Signify ones are at the new server.

bluemoehre, 2024-10-27:
@ebaauw I updated the screenshots

ebaauw, 2024-10-28:
Thanks.  This is a ZLL light, so I think it can be added to `light_zll_C.json`.

bluemoehre, 2024-10-28:
@ebaauw I did some research and it looks like there are various generations of those lights. Some don't support effects at all (like this one), some only a few, and the most recent generation (see my other device request) from 2021 supports all.

As these differences between generations also affect some other models - how do we reflect this in the DDF? Is this why we have *A, *B, *C files?

ebaauw, 2024-10-29:
The light reports what effects it supports, so no need to differentiate for that in the DDFs.  ZLL lights do offer fewer features than ZB3 lights (notably attribute reporting), so they do need different DDFs.  Gamut A vs B vs C indicate what colours the light support (or rather what RGB LEDs it contains).  While this is reported by the light as well, there’s differences in other features (like no `ct` for gamut A) that do require different DDFs.

closed 2024-11-04 by bluemoehre