dresden-elektronik/deconz-rest-plugin #7997: Philips Hue Play Lightbar (440400982842)

bluemoehre, 2024-10-26:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Hue Play Lightbar

### Manufacturer

Philips

### Model identifier

440400982842

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]
[image]

### Color Control

[image]
[image]
[image]
[image]
[image]
[image]

### HUE Entertainment

[image]

ebaauw, 2024-10-26:
You need to update the light firmware, so it can be added to one of the existing DDFs for Hue lights.  What’s the _Image Type_ of this light?

bluemoehre, 2024-10-26:
The firmware is latest. I updated the devices right before posting the screenshots here.

Image Type: `0x01001d0c`

ebaauw, 2024-10-27:
Latest 0x011f firmware is 0x01002402 (16786434), for _SW Build ID_ 1.222.2.

Sometimes firmware updates are staged, requiring multiple successive upgrades to get to the latest.  I've seen this for some Hue lights, that used to have another _Image Type_ before changing to 0x011f.

Also, make sure to re-read in the GUI the _Simple Descriptors_ in the _Basic_ cluster attributes.  The light should advertise an FC03 _Hue Effects_ cluster.

bluemoehre, 2024-10-27:
> Sometimes firmware updates are staged, requiring multiple successive upgrades to get to the latest

That's interesting. The device was listed as "up-to-date", but after repairing via BT to the HUE app I could update again. Still it's not the latest version. Gonna check how often I need to do this. Since I own the device multiple times, I will try updating another via deCONZ directly to latest and see if that works as well.

bluemoehre, 2024-10-27:
@ebaauw I updated the screenshots

ebaauw, 2024-10-28:
To be added to `light_zb3_C.json`.