dresden-elektronik/deconz-rest-plugin #7999: humidity, temperature sensor TUYA is not getting the battery

SvenMartin81, 2024-10-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hi,
Phoscon is not getting the battery from this sensor: _TZE200_9yapgbuv - Product: TS0601 - Version 1.0.8
 

### Steps to reproduce the behavior

it is always

### Expected behavior

i should have the status of the battery as I have it for other sensors "_TZ3000_saiqcn0y" - Product: TS0201 - Version 1.0.6

### Screenshots

[image]
[image]

### Environment

- Host system: (Raspberry Pi 4b)
- Running method: (Raspbian )
- Firmware version: (26780700)
- deCONZ version: (2.29.0)
- Device: ( RaspBee II)
- Do you use an USB extension cable: (no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? no

Smanar, 2024-10-29:
The model "_TZ3000_saiqcn0y" use the DDF _TZ3000_TS0201_temp_hum_sensor.json (use classic cluster)
The model "_TZE200_9yapgbuv" use the DDF _TZE200_TS0601_humidity_temp.json (use tuya cluster)

But I have checked again it seem it use the good dpid (the 0x04)

It's strange there is no value at all, perhaps the value is still "none", can you check direclty in the API ? using Phoscon/help/API Information/sensors.

Else no other way than checking logs if the device don't use another dpid.

SvenMartin81, 2024-10-29:
Hi, 
I am using the Beta Gateway verion 2.29.0.
By looking into the DDF file you are menshen, i can't my version "_TZE200_9yapgbuv". could this be the reason?

I deleted this sensor and learnd it again to the system. but there was no change in the Battery-sensor.....it is still not availible.

Smanar, 2024-10-30:
>By looking into the DDF file you are menshen, i can't my version "_TZE200_9yapgbuv". could this be the reason?

No I m seing you can't compare yours 2 device, they are not working same.
I have see that https://github.com/zigpy/zha-device-handlers/issues/2281

This device don't use the same dpid, If I have understand It use the dpid = 3 and return 3 value for low/medium/high

Can you edit the DDF to have 

```
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

You have it at 2 position in the DDF, then restart deconz, if it work you will have value 0 (for 25%), 1 for (50%) and 2 (for 100%)

Smanar, 2024-10-31:
Yes, It's exaclty that, what is your OS ? not docker ?
Easier to just edit the file, change the dpid, save and restart deconz.

You can edit direclty the file, it will have an impact on all thoses devices
`  "manufacturername": ["_TZE200_nnrfa68v","_TZE200_znbl8dj5", "_TZE200_qoy0ekbd", "_TZE200_a8sdabtg", "_TZE200_whkgqxse","_TZE200_yjjdcqsq", "_TZE200_9yapgbuv", "_TZE200_utkemkbs", "_TZE200_locansqn", "_TZE204_yjjdcqsq"],`

But the rollback is easy, and on next deconz update, the file will be reseted. It's just to test.

SvenMartin81, 2024-10-31:
Hey,

I changed it to 3. but it is the Same as before

Smanar, 2024-11-01:
Can you check the value direclty in the API ?
Using Phoscon/help/API Information/sensors ?

With some luck the value is 0,1 or 3.

SvenMartin81, 2024-11-01:
Hi,

Batteris ist "1" but now I can see the batterie in the sensor but with 1%

                    {
    "config": {
        "battery": 1,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "641473d73f6d2509c6a40d4fe0c8ebe4",
    "lastannounced": null,
    "lastseen": "2024-11-01T14:34Z",
    "manufacturername": "_TZE200_9yapgbuv",
    "modelid": "TS0601",
    "name": "Temp Feuchtigkeit Gewächshaus",
    "state": {
        "humidity": 6600,
        "lastupdated": "2024-11-01T14:34:10.810"
    },
    "swversion": "1.0.8",
    "type": "ZHAHumidity",
    "uniqueid": "a4:c1:38:78:ab:a5:f1:38-01-0405"
}
                    
                
[image]

Smanar, 2024-11-02:
No it's normal 

>You have it at 2 position in the DDF, then restart deconz, if it work you will have value 0 (for 25%), 1 for (50%) and 2 (for 100%)

Was a test, and a godd new, the complete working code can be

```
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 3, "if (Attr.val == 0) { Item.val =25 } else if (Attr.val == 1) { Item.val = 50 } else { Item.val =100 }"},
          "read": {"fn": "none"},
          "default": 0
        },
```

SvenMartin81, 2024-11-02:
Hi I changed the DDF in the battery area to this:
       {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 3, "if (Attr.val == 0) { Item.val=25 } else if (Attr.val == 1) { Item.val=50 } else { Item.val=100 }" },
          "read": {"fn": "none"},
          "default": 0
        },

but nothing changes after a restart from deconz....

SvenMartin81, 2024-11-02:
Hi I changed the DDF in the 2 x battery area to this --> i thought there was a missing part "eval":
{
"name": "config/battery",
"parse": {"fn": "tuya", "dpid": 3, "eval": "if (Attr.val == 0) { Item.val=25 } else if (Attr.val == 1) { Item.val=50 } else { Item.val=100 }" },
"read": {"fn": "none"},
"default": 0
},

but nothing changes after a restart from deconz.... The only difference is that I can see the battery again in the phoscon app but only 1%.

SvenMartin81, 2024-11-03:
Hi, 
I just look into the Sensor in Phoscon and you were Right. With Code from yesterday, the Battery is on 50%. Should I do something Else. I am not sure if the Battery is realy on 50%

SvenMartin81, 2024-11-03:
> Hi I changed the DDF in the 2 x battery area to this --> i thought there was a missing part "eval": { "name": "config/battery", "parse": {"fn": "tuya", "dpid": 3, "eval": "if (Attr.val == 0) { [...]

This is my actually Code.

Smanar, 2024-11-03:
>Hi,
I just look into the Sensor in Phoscon and you were Right. With the Code from yesterday, the Battery is on 50%. Should I do something Else. I am not sure if the Battery is realy on 50%

No for me it's fine, value are from Z2M.

But need to make a DDF just for this device (as other are working with old code)

Do you want to make a PR to sumbit it or want I make it ? (need to remove the device from the old DDF and create a new one just for it)

SvenMartin81, 2024-11-03:
Hi, this should be the file.

[_TZE200_TS0601_9yapgbuv_humidity_temp.json](https://github.com/user-attachments/files/17611830/_TZE200_TS0601_9yapgbuv_humidity_temp.json)

closed 2024-11-08 by manup