dresden-elektronik/deconz-rest-plugin #8000: Info: deCONZ-autostart.sh "$OS_ID" = "raspbian/debian"

ltspicer, 2024-10-27:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

deconz-gui will not be restartet.

### Steps to reproduce the behavior

1. Close deCONZ-gui (VNC)
2. Reopen deCONZ-gui

### Expected behavior

deCONZ-GUI should start again.

### Screenshots

[image]

### Environment

- Host system: Raspberry Pi 5
- Running method: Raspberry OS /  ioBroker / deconz-gui
- Do you use an USB extension cable: yes -- ConBee III
- Additional USB3 connected with SSD

### Additional context

deCONZ-autostart.sh
In the lines 29 and 101 in the if statement is "$OS_ID" = "raspbian"
For Raspberry OS it does be "$OS_ID" = "debian"

SwoopX, 2024-11-18:
@manup This is a topic for deconz core. Can we maybe move that over (I can't)?

manup, 2024-11-21:
The deCONZ repository didn't have the issue tracker activated yet. I've activated this now, and transfer this issue. We need to keep an eye on these to proper sort them to the right repos.