dresden-elektronik/deconz-rest-plugin #8005: IKEA TRÅDFRI bulb E27 WS clear 806lm

bluemoehre, 2024-10-30:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

TRÅDFRI bulb E27 WS clear 806lm

### Manufacturer

IKEA of Sweden

### Model identifier

TRÅDFRI bulb E27 WS clear 806lm

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]

### Color Control

[image]
[image]
[image]

mattreim, 2024-10-31:
And here change the “modelid”, “product” and the name of this json. Then it should work and the intervals should be checked for all IKEA DDF's.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/tradfri_bulb_e14_ws_opal_600lm.json

bluemoehre, 2024-10-31:
> and the intervals should be checked for all IKEA DDF's.

You are right. Some attributes are polled every `5s`. That will cause spamming the network, am I wrong? Seems to be a larger issue and related to a bunch of devices. Therefore that topic should have a dedicated ticket.

mattreim, 2024-10-31:
It's true, but what's more important is that it works.

ebaauw, 2024-11-01:
I think all IKEA devices I’ve seen support attribute reporting.  Better use that instead of polling.

bluemoehre, 2024-11-23:
Waiting for other PRs to get merged before continuing here.

mattreim, 2025-02-17:
Is merged [tradfri_bulb_e27_ws_clear_806lm.json](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/tradfri_bulb_e27_ws_clear_806lm.json) and the issue can be closed.

closed 2025-02-18 by bluemoehre