dresden-elektronik/deconz-rest-plugin #8006: IKEA TRÅDFRI bulb GU10 W 400lm

bluemoehre, 2024-10-31:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

IKEA TRÅDFRI bulb GU10 W 400lm

### Manufacturer

IKEA of Sweden

### Model identifier

TRADFRI bulb GU10 W 400lm

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]

### IKEA Specific

[image]

bluemoehre, 2024-10-31:
Maybe someone can check if the existing model is the same and/or what is the difference causing no DDF match:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/tradfri_bulb_gu10_w_400lm.json#L5

mattreim, 2024-10-31:
Try changing the status from silver to gold.

bluemoehre, 2024-11-23:
Waiting for other PRs to get merged before continuing here.

mattreim, 2025-02-15:
Is merged [tradfri_bulb_gu10_w_400lm.json](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/tradfri_bulb_gu10_w_400lm.json)

mattreim, 2025-02-17:
Is merged [tradfri_bulb_e27_ws_clear_806lm.json](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/tradfri_bulb_e27_ws_clear_806lm.json) and the issue can be closed.