dresden-elektronik/deconz-rest-plugin #8013: The write function is ignored defined by DDF for window covering device

LukasA83, 2024-11-02:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have a Merten Windows covering device, which is function great. The position is reversed (0 is closed). I applied a fix in the ddf, see below. It works for reading the values, but not for sending the values via REST API.

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "1GANG/SHUTTER/1",
  "vendor": "Schneider Electric",
  "product": "Merten Wiser cover controller (MEG5113-0300)",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/meg5113-0300_cover_controller_custom.json",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x05"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lift",
          "refresh.interval": 300,
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "write": {
            "cl": "0x0102",
            "cmd": "0x05",
            "ep": 0,
            "eval": "100-Item.val;",
            "fn": "zcl:cmd"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 5,
            "eval": "Item.val = Attr.val == 255 ? 50 : (100 - Attr.val);",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "state/open",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 5,
            "eval": "Item.val = Attr.val != 0;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 5,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}`

### Steps to reproduce the behavior

1. Apply DDF
2. Read "lift" attribute via Rest API -> Works e.g. 51
3. Write "lift:51" attribute via Rest API -> The command works
4. Read "lift" attribute via Rest API -> Fail, indicates 49

### Expected behavior

DDF write expression is applied when sending values via RestAPI.

### Environment

- Host system: (Raspberry Pi )
- Running method: (deconz-docker container)
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.29.0)
- Device: (RaspBee II)

Smanar, 2024-11-02:
Yes, it's a know issue, it's for that the support for covering that use the tuya cluster is not possible.

LukasA83, 2024-11-02:
Hi Smanar,

Thanks for prompt response. It is not a Tuya device / has a Tuya cluster?!

closed 2024-12-02 by github-actions[bot]