dresden-elektronik/deconz-rest-plugin #8020: Sonoff ZBMINIR2

duffbeer2000, 2024-11-05:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

SONOFF ZBMINI Extreme Zigbee Smart Switch (Neutral Wire Required) | ZBMINIR2

### Manufacturer

Sonoff

### Model identifier

ZBMINIR2

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

### OTAU
[image]

### FC11
[image]

### FC57
[image]

duffbeer2000, 2024-11-05:
A little bit more information:
- I tried to create a DDF out of the ZBMINIL2 --> Works but is not complete

Difference between ZBMINIR2 to ZBMINIL2 and zbmini:
R2 is a router, the L2 is an end device
[image]

So what's missing:
- External Switch (missing vor the zbminil2 too)
- Detach Relay Mode
- Power-on State (missing vor the zbminil2 too)

What I found out:
- The FC11 Cluster is not complete
- 0x0014 --> Power-On State (on/off)
- 0x0015 --> Power-On State Tme (time till it goes off (0-59min)
- 0x0016 --> External Switch (0 = edge, 1 = pulse, 2 = following (off), 130 = following (on)
- 0x0017 --> Detach Relay Mode (enabled = 1, disabled = 0)

Smanar, 2024-11-05:
>R2 is a router, the L2 is an end device

Yes, but I still thinking it's not a sleeping device, so don't worry for that.

For the rest you can use for exemple

```
        {
          "name": "config/devicemode",
          "refresh.interval": 3000,
          "read": {
            "at": "0x0017",
            "cl": "0xfc11",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0017",
            "cl": "0xfc11",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'disabled') { 0 } else if (Item.val == 'enabled') { 1 }",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0017",
            "cl": "0xfc11",
            "ep": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'disabled' } else { Item.val = 'enabled' }",
            "fn": "zcl:attr"
          },
          "default": "disabled"
        },
```

But you need
- A ZHA sensor, on ZHAlight the "config" can be hidden
- Find a field to use, here I m using "config/devicemode", but you need more (it's possible to create new one)
- Take care at dt, on this code I m using `"dt": "0x20",` but if this attribute is a bool, need to use "dt": "0x10",

duffbeer2000, 2024-11-07:
I think that's to high for me. ;) I can't create a working DDF.
But what I was able to do is to edit general.xml so I'm able to get and set the values there. What's not working there are the value substitution with dropdown. Maybe you can tell me what I did wrong,

```
<!-- SONOFF -->
<cluster id="0xfc11" name="Sonoff specific" mfcode="0x1286">
  <description>Cluster for Sonoff Specific Attributes</description>
  <server>
	<attribute id="0x0012" name="Radio Power Turbo Mode" type="s16" access="r" required="m">
	  <value name="False" value="0x09"></value>
	  <value name="True" value="0x14"></value>
	</attribute>
	<attribute id="0x0014" name="Power-On State" type="bool" access="rw" required="m"></attribute>
	<attribute id="0x0015" name="Power-On Time" type="u16" access="rw" required="m"></attribute>
	<attribute id="0x0016" name="External Switch Mode" type="u8" access="rw" required="m">
	  <value name="Edge" value="0x00"></value>
	  <value name="Pulse" value="0x01"></value>
	  <value name="Following Off" value="0x02"></value>
	  <value name="Following On" value="0x89"></value>
	</attribute>
	<attribute id="0x0017" name="Detach Relay Mode" type="bool" access="rw" required="m"></attribute>
	<attribute id="0x2001" name="Brightness" type="u8" access="r" required="m" mfcode="0x1286"></attribute>
  </server>
  <client>
  </client>
</cluster>
```

[image]

`
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "ZBMINIR2",
  "vendor": "Sonoff",
  "product": "Mini Relay (ZBMINIR2)",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/zbminir2.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5,
          "read": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 0,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/devicemode",
          "parse": {
            "at": "0x0017",
            "cl": "0xfc11",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 1800
        }
      ]
    }
  ]
}`

Smanar, 2024-11-07:
Try with
```

	<attribute id="0x0016" name="External Switch Mode" type="enum8" access="rw" required="m">
	  <value name="Edge" value="0x00"></value>
	  <value name="Pulse" value="0x01"></value>
	  <value name="Following Off" value="0x02"></value>
	  <value name="Following On" value="0x89"></value>
	</attribute>
```

duffbeer2000, 2024-11-07:
And for the DDF i posted after the Screenshot of the cluster?

duffbeer2000, 2024-11-08:
DDF: It took me hours to find the error and now I finally found it. "config/devicemode" is a string and I had a bool! Till I switched to "config/configured" my first tests are working. I'll try to complete it later.
How can I create new fields and load it to use it. I tried to place it under devices/general/items but after editing the DDF and restart of deConz it don't show up in the Rest API and it got removed from the ddf and I see in the logs:
15:09:04:045 Mime-format: text/uri-list
Mime-data: ddfitem:config/detachrelaymode

Button press:
I searched long time to find a buttonpress event in the logs but the only thing I found in the logs is that these 5 entries are always and only written when I change the state of the edge switch:
13:46:50:302 APS-DATA.indication srcAddr: 0x54E3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 179, rssi: -77
13:46:50:304 APS-DATA.request id: 201, addrmode: 0x02, addr: 0x54E3, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x04
13:46:50:414 APS-DATA.confirm id: 201, status: 0x00 SUCCESS
13:46:50:415 APS-DATA.confirm request id: 201 -> erase from queue
13:46:50:489 aps request id: 201 finished, erase from queue

Smanar, 2024-11-08:
>How can I create new fields and load it to use it. I tried to place it under devices/general/items but after editing the DDF and restart of deConz it don't show up in the Rest API and it got removed from the ddf and I see in the logs:

It's exactly that. Perhaps a typo ? If I have understand what you are trying to do

config_detachrelaymode_item.json

```
{
  "schema": "resourceitem1.schema.json",
  "id": "config/detachrelaymode",
  "datatype": "Bool",
  "access": "RW",
  "public": true,
  "description": "XXXXXXXXXXx."
}
```

Then after that you can use it in DDF after a deconz restart.

But if you create too much new field, it will be hard to validate your DDF (big debate every time), so try at maximum to use existing one.

For buttonevent you need to have logs like this one
`
[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP_WITH_ON_OFF (0x06) payload[0]: 001`

Else if you realy have nothing in the buttonmap file
`[INFO] - No button handler for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n`

duffbeer2000, 2024-11-10:
So I tried a lot (didn't know I had to add the clusterID to the uuid of the switch) and I think I'm 75% ready with the ddf. I created a fork with my changes:
[https://github.com/duffbeer2000/deconz-rest-plugin](https://github.com/duffbeer2000/deconz-rest-plugin)

### What's missing: 
- radioPower: {Cluster 0xfc11, Attribute ID: 0x0012, type: INT16}, 'Enable/disable Radio power turbo mode', valueOff: [false, 0x09], valueOn: [true, 0x14], --> Maybe I need another config item.
- config/on/delayedpoweron --> Missing in the API output
- config/on/startup --> Missing in the API output
- The Buttons are not working correctly at the moment

### Switches
With Detached Relay active the Button is working but I don't get a new last updated from the API
ButtonMap for Detached Relay:
[1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "short"]
Result: 
20:46:08:736 [INFO] - Button 1002 - ZBMINIR2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: short, payload: None, zclSeq: 116

Rocker Switch 
On  - 18:24:19:817 [INFO] - No button map for: ZBMINIR2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001001, zclSeq: 100
Off - 18:24:24:789 [INFO] - No button map for: ZBMINIR2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001000, zclSeq: 101

With the following button map it's not working:
[1, "0x01", "ONOFF", "ATTRIBUTE_REPORT", "0x01", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press 1"],
[1, "0x01", "ONOFF", "ATTRIBUTE_REPORT", "0x00", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press 2"],

Result:
ON: 21:05:57:150 [INFO] - Button 1002 - ZBMINIR2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Normal press 1, payload: 00001001, zclSeq: 124
Off: 21:06:02:112 [INFO] - Button 1002 - ZBMINIR2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Normal press 1, payload: 00001000, zclSeq: 125

### API Output
That's the output of the API at the moment with http://192.168.178.152:8189/api/XXXXXX/devices/D4:48:67:FF:FE:3B:8F:DF
{
    "ddf_policy": "latest_prefer_stable",
    "lastannounced": null,
    "lastseen": "2024-11-10T16:13Z",
    "manufacturername": "SONOFF",
    "modelid": "ZBMINIR2",
    "name": "On/Off output 10",
    "productid": "ZBMINIR2",
    "subdevices": [
        {
            "state": {
                "on": {
                    "lastupdated": "2024-11-10T13:46:46Z",
                    "value": false
                },
                "reachable": {
                    "lastupdated": "2024-11-10T17:05:41Z",
                    "value": true
                }
            },
            "type": "On/Off output",
            "uniqueid": "d4:48:67:ff:fe:3b:8f:df-01"
        },
        {
            "config": {
                "clickmode": {
                    "lastupdated": "2024-11-10T13:48:46Z",
                    "value": "rocker"
                },
                "delay": {
                    "lastupdated": "2024-11-10T00:02:18Z",
                    "value": 0
                },
                "detachrelaymode": {
                    "lastupdated": "2024-11-10T13:48:03Z",
                    "value": true
                },
                "on": {
                    "lastupdated": "2024-11-10T14:38:02Z",
                    "value": true
                },
                "reachable": {
                    "lastupdated": "2024-11-10T17:00:24Z",
                    "value": true
                }
            },
            "mode": 1,
            "state": {
                "buttonevent": {
                    "lastupdated": "2024-11-10T00:11:08Z",
                    "value": 1002
                }
            },
            "type": "ZHASwitch",
            "uniqueid": "d4:48:67:ff:fe:3b:8f:df-01-0006"
        }
    ],
    "swversion": "1.0.2",
    "uniqueid": "d4:48:67:ff:fe:3b:8f:df"
}

Smanar, 2024-11-11:
>With Detached Relay active the Button is working but I don't get a new last updated from the API

The state/buttonevent on the ZHASwitch is not updated ?

duffbeer2000, 2024-11-11:
In the API I always get this, I guess it's because in detached realy mode it always get's a 1002 and so it doesn't get updated but maybe I'm wrong :

> "state": {
> 	"buttonevent": {
> 		"lastupdated": "2024-11-10T00:11:08Z",
> 		"value": 1002
> 	}
> },

Smanar, 2024-11-11:
Ha, yes, it's possible you have always the same value.
But you will have a new notification at every button press.

duffbeer2000, 2024-11-12:
I have a little problem with the ddf. I triy to write the value back with the API but it don't work and I don't know why, did a lot of try and error with it. With the API  I cannot set the value, even with off which responds success it don't set it in the device.

Response for off:
[ { "success": { "/sensors/18/config/mode": "off" } } ]

Response for on:
[ { "error": { "address": "/sensors/18/config/mode", "description": "Could not set attribute", "type": 608 } } ]

"name": "config/mode",
"description": "Enables or disable the Turbo Mode to Boost the signal.",
"write": {
  "at": "0x0012",
  "cl": "0xfc11",
  "dt": "0x29",
  "ep": 1,
  "eval": "if (Item.val == 'off') { 9 } else if (Item.val == 'on') { 32 }",
  "fn": "zcl:attr"
},

With the Attribute Editor it works:
[image]

Smanar, 2024-11-13:
It's probably a "security" have you added 

```
      "meta": {
        "values": {
          "config/mode": {"off": 9, "on": 32}
        }
      },
```

duffbeer2000, 2024-11-13:
No I don't have meta. Saw it in other DDFs but didn't know what it means. I'm not at home till tomorrow an will test that tomorrow evening.
To be sure, valueOff: [false, 0x09] is 9 and valueOn: [true, 0x14] is 32? Am not sure if I make that right.

Smanar, 2024-11-14:
32 is 0x1f 
0x14 is 20

duffbeer2000, 2024-11-14:
Ok, Put on and off with the API gives a success but the values never reach the device and I don't know why.
"write": {
  "fn": "zcl:attr",
  "ep": 1,
  "cl": "0xfc11",
  "at": "0x0012",
  "dt": "0x29",
  "mf": "0x1286",
  "eval": "if (Item.val == 'off') { 9 } else if (Item.val == 'on') { 20 }"
},

duffbeer2000, 2024-11-15:
Found it, added the mf code for a test and forgot to remove it for the tests with the meta attribute. Without the manufactor code it's working.

duffbeer2000, 2024-11-15:
> Ok, with meta Put on and off with the API gives a success but the values still never reach the device and I don't know why. "name": "config/mode", "write": { "fn": "zcl:attr", "ep": 1, "cl": "0xfc11", "at": "0x0012", "dt": "0x29", "mf": "0x1286", "eval": "if (Item.val == 'off') { 9 } else if (Item.val == 'on') { 20 }" },

@Smanar can you help me with that, too?

AbDhops, 2024-11-15:
Any thoughts on when this will be part of official release?

duffbeer2000, 2024-11-15:
At the moment we are at 95% (I hope so, it's my first DDF)
https://github.com/duffbeer2000/deconz-rest-plugin (Smanar made a fork for full support of the button events of that device. So that one is needed too if for on and off.) 

What's missing:
- config/on/delayedpoweron --> Missing in the API output (new config item, which hopefully will get added)
If you set config/on/startup to true you can set Delayed Power-On (on/off) which means after a compelte power loss in the building you can define a time delay when power is back. So not all devices switch on the same time. To activate that feature the config item is needed.

I think it has to be added in the resource.cpp, resource.h and rest_lights.cpp.

@Smanar can you help me with the config Item? And do you think I have a chance that the config item get's added?

AbDhops, 2024-11-16:
Okay.
I am running the deconz in container (version `deconzcommunity/deconz:2.29.0 `).
My device is succesfully paired. However it is operating individually from app and from switch. So I turn on or off from Switch, I am not getting status in Deconz. 

Should I open new issue on that, or will above fixes will resolve those ? If require, I can try beta version and help with logs.

duffbeer2000, 2024-11-16:
I already added it under devices\generic\items\config_on_startup_item.json but I don't get it in the rest response.

Smanar, 2024-11-16:
I don't see the problem, I m asking to others devs

Smanar, 2024-11-16:
You have a typo here

```
          "refresh.interval": 5,
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0006",
            "at": "03x0000"
          },
```

On read part, the attribute.
And better to set bigger value for the refresh, bigger than the one set on bind/report.

duffbeer2000, 2024-11-16:
Switchet to:
{
  "name": "state/on",
}
Should I change the refresh from the native one to something bigger? 10 or higher?

Smanar, 2024-11-16:
Lol, I think you have found a leak.
When you make a PR with a DDF there is a bot that check thoses value, and the DDF is rejected if the refresh is bigger than value used on bind.
Except if you use the defaut value ^^.

And no, all the code is working with this value, better to let it.

catartica76, 2024-11-20:
What is the timeframe for ZBMINIR2 support on deCONZ?
could it make it into v2.29.0 stable?

thanks

Smanar, 2024-11-21:
I don't see timeframe, so I realy don't have idea.
Can wait at least for devs reaction, this code need c++ modification so they will be more drastic.

duffbeer2000, 2024-12-17:
Push

Smanar, 2025-01-08:
The DDF is working but still in waiting list.

gitfvb, 2025-01-13:
Just installed a sonoff zbminir2 and it worked straight away with 2.27.6

Just used Phoscon and searched for new lights.

You can see it here in the screenshots. [image] [image]

closed 2025-02-11 by github-actions[bot]

sciorty, 2025-03-07:
I just added mine, I can control it but when I manually switch it on/off the state is not reported in phoscon. Why is that? I tried to factory reset it but didn't change.

[image]