dresden-elektronik/deconz-rest-plugin #8023: Hive Thermostatic Radiator Valve (Danfoss TRV003)

Sergeus, 2024-11-06:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Hive Smart Heating Thermostatic Radiator Valve

### Manufacturer

danfoss

### Model identifier

TRV003

### Device type to add

Thermostat

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Sergeus, 2024-11-06:
I think this device is a model variant of the existing danfoss TRV:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/danfoss/etrv0100_thermostat.json

But presumably since the model ID (TRV003) doesn't match any in there, it isn't picked up atm?

Smanar, 2024-11-07:
It's easy to test, just add your model id + Manufacture name on the actual DDF.
Then restart deconz and try to include it.

Mimiix, 2024-11-07:
Please add all relevant screenshots

SwoopX, 2024-11-07:
> Please add all relevant screenshots

@Mimiix In this particular case, this is not needed. Danfoss has their cluster documentation publicly available.

Looks all like it's just another manufacturername/modelid combo. Remarkable though is the firmware version, the latest one all my thermostats have is 1.28, here it is 2.50 🤔

Sergeus, 2024-11-08:
> It's easy to test, just add your model id + Manufacture name on the actual DDF.
> Then restart deconz and try to include it.
Awesome, I'll have to try this later this weekend. I'll need to figure out how to actually access the ddf file in the layers of VMs and containers I've got running!

> Please add all relevant screenshots 
For my future reference, which screenshots are missing? From SwoopX's comment I suspect it's something to do with clusters. I tried to follow the docs for instructions on the screenshots to include but I was least sure about my understanding of the cluster part

Smanar, 2024-11-08:
The usefull interesting missing cluster is the thermostat one (and configuration one to see wich one attribute to use for battery).

>I'll need to figure out how to actually access the ddf file in the layers of VMs and containers I've got running!

What is your OS ? on VM it's easy, on docker, depend of your OS.