dresden-elektronik/deconz-rest-plugin #8027: BOSCH Thermostat II (RBSH-TRV0-ZB-EU) cannot be connected to the Conbee III (deconz)

Dermayer, 2024-11-08:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

When teaching in the Bosch thermostat, I have so far sent the Zigbee EUI via Node red and the installation code via the REST API. This was possible with the Conbee 2 stick without any problems. This is no longer possible with the Conbee 3

### Steps to reproduce the behavior

Set the Bosch thermostat to connection mode
Phoscon on pairing devices
Send the installation code via Node Red

### Expected behavior

That the thermostat is integrated as with the Conbee 2

### Environment

- Host system: Raspberry Pi 4b
- Running method: Raspbian 
- Firmware version: (26530900)
- deCONZ version: (2.28.1)
- Device:  ConBee III
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so:  64GB SSD

Mimiix, 2024-11-08:
Are other devices working as expected?

Dermayer, 2024-11-09:
Yes, around 40 different devises work without any problems. After pairing the Bosch thermostats with the Conbee 2, they also work without problems with the Conbee 3

NetSpider-AT, 2024-11-19:
I have the same problem. The existing thermostat was online in the GUI and the 'last seen' field was updated, but it did not respond to changes. The thermostat itself was no longer connected. After deleting and resetting the device, it can no longer join. A new thermostat also cannot join.

chrissiboy, 2024-12-04:
I can confirm this behavior (Conbee III) at Ubuntu environment with Deconz 2.29.1. No problems with Conbee II.

closed 2025-01-02 by github-actions[bot]

Dermayer, 2025-01-03:
Please reopen, the problem still exists for me. @chrissiboy, @NetSpider-AT : What about you?

F3LIX95, 2025-01-20:
Trying for hours now with a new Compact Plug EU on V2.28.1. Provided InstallCode via API. It won't pair... Unfortunately only have a Conbee III

Dermayer, 2025-02-13:
I had to reset a BOSCH Thermostat II. The integration was again only possible via the ConBee II stick. 
Absloult no change with ConBee III

Please open issue again.

krppelgewchs, 2025-12-31:
Please open the issue again and comment on current status, if possible.