dresden-elektronik/deconz-rest-plugin #8037: ION Industries LED Zigbee Dimmer (ID200W-ZIGB)

theNailz, 2024-11-22:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

LED Zigbee Dimmer (ID200W-ZIGB)

### Manufacturer

ION Industries

### Model identifier

TS0601

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

## Clusters
[image]

## Identify
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]
[image]

### Level Control
[image]
[image]

### Any other cluster of relevance/interest

## Groups
[image]

## Scenes
[image]

## Level Control
[image]

## OTAU
[image]

## Touchlink Commisioning
[image]

theNailz, 2025-01-08:
Preemptive bump

theNailz, 2025-01-29:
If there is any way I could try to get it working myself, I'd love to learn how. Are there any development resources available?

Smanar, 2025-02-16:
Hello, your model is a tuya device, and use Tuya cluster, so you can forget all standard.
It's a clone of devices (for exemple) https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/moes/Moes_MS-105-M_1_gang_dimmer_module.json

Can try with adding your model (then restart deconz)

But this device don't use the "tuya unlock sequence", can need it for some device
```

        {
          "name": "config/tuya_unlock"
        },
```

theNailz, 2025-02-16:
@Smanar Cool, this made it work, but: Every time you change the brightness, the light fades from 0 to the desired brightness. So if the light is at 80% and you set it to 100%, it goes 80% -> 0% -> fade to 100%, all in under a second. This causes a flicker effect when using it. (How) can I solve this?

Smanar, 2025-02-17:
He ? Good question, I realy don't see where it can come.
Try with taking a look on logs with flag" DDF" or both "Info", you will see request made by deconz

If all is fine deconz make only 1 request, with the requested value (so no request to return to 0 value)

closed 2025-03-18 by github-actions[bot]