dresden-elektronik/deconz-rest-plugin #8040: Cannot turn off siren - Frient Intelligent smoke alarm (SMSZB-120)

SvenNilsson, 2024-11-24:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

The Frient Intelligent smoke alarm should support setting 3 states via the rest apis on the "warning device" endpoint:

- https://deconz.archaedoit.se/api/<api-key>/lights/<id>/state.
- body { "alert": "select" } and { "alert": "lselect" } and  { "alert": "none" }

select and lselect would turn on the siren and none should turn it off.

sending select and none does nothing. (sending lselect, turn on siren which has to be manually reset/turned off by pressing the button on the smoke alarm (front))

### Steps to reproduce the behavior

use put command on endpoint:
https://deconz.archaedoit.se/api/<api-key>/lights/<id>/state

first with body: { "alert": "select" }, siren does not go off

then with body: { "alert": "lselect" } siren goes off

then with body  { "alert": "none" } siren does not stop

### Expected behavior

 { "alert": "none" }  stops the siren
 { "alert": "select" } start the siren

### Screenshots

[image]

### Environment

- Host system: Synology NASwith docker
- Running method: docker: deconzcommunity/deconz:latest 
- Firmware version: 26780700
- deCONZ version (not Home assistant Addon version!): 2.29.0 and 2.28.1
- Device: ConBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? no
-Frient smoke alarm has firmware 4.0.8 (latest from what I could tell)

### Additional context

Form what I could see reading the code the DDF have been changed lately (August 8) and actually tested with my device... So it could have something to do with this part of the DDF, if this issue didn't exist before:

{
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "Start warning command payload has 5 Bytes. Byte 1: Options. 0x17 = Warning mode 1 (burglar), Strobe, Very high sound. Byte 2-3: Duration, Byte 4: Strobe duty cycle, Byte 5: Strobe level",
          "read": {
            "fn": "none"
          },
          "write": {
            "cl": "0x0502",
            "cmd": "0x00",
            "ep": "0x01",
            "eval": "if (Item.val=='select') { '1701000000' } else if (Item.val=='none') { '0000000000' }",
            "fn": "zcl:cmd"
          },
          "default": "none"
        },
        {
          "name": "state/reachable"
        }
      ]

[image]

SvenNilsson, 2024-12-08:
For anybody reading this and have the same issues, using the previous ddf file solved the issue for me. Seems like this line introduces the bug: "eval": "if (Item.val=='select') { '1701000000' } else if (Item.val=='none') { '0000000000' }",

SwoopX, 2025-01-26:
@SvenNilsson 
Hm, this is indeed a "bug". I seem to have missed to set the correct endpoint for the command in the previous PR. Mea culpa. Changing the `ep` value to `35` should already do the trick.

Sorry, but I currently can't test is myself.

SvenNilsson, 2025-01-27:
Hi @SwoopX,
I'm happy to test it, just let me know how. (currently running this as docker container).

SwoopX, 2025-01-27:
@SvenNilsson  Would be good if you could do so. Like said, I can't do it myself anytime soon.

pawlizio, 2025-03-25:
Have the same issue! Would appreciate to have this fixed soon.

pawlizio, 2025-03-25:
@SwoopX Thanks for the hint, setting the ep value to 35 solved the issue.

pawlizio, 2025-03-25:
By the way, I think there is also an issue with the state when the siren is switched on, the switch automatically turns off (in Home Assistant) even though the alarm is on. 

Not sure if the root cause is in this library or probably on HA side: 
https://github.com/Kane610/deconz/blob/5ce2b6307572e7720e80628aca4b38fbf156661f/pydeconz/models/light/siren.py#L27-L29

closed 2025-03-26 by manup

mariomaz87, 2025-04-18:
> [@SwoopX](https://github.com/SwoopX) Thanks for the hint, setting the ep value to 35 solved the issue.

Hi, I'm trying to fix this as well. Where can I set the EP value? I can't find in using the VNC interface.

pawlizio, 2025-04-18:
I created a modified ddb/ddf file using [deconz toolbox](https://deconz-community.github.io/ddf-tools/#/auto-bundler) and uploaded it using the phoscon webinterface. 

However you do not need to change the EP value. You can simply download the latest (already modified) ddb/ddf [here](https://deconz-community.github.io/ddf-tools/#/store/bundle/59df07817b49eceee04f77616e05b179e22b31d6aa51b009a0b95fc832abc612) and load it to your phoscon instance, following this description: https://phoscon.de/en/app/doc#device-integration-en.

mariomaz87, 2025-04-18:
> I created a modified ddb/ddf file using [deconz toolbox](https://deconz-community.github.io/ddf-tools/#/auto-bundler) and uploaded it using the phoscon webinterface.
>
> However you do not need to [...]

Thanks. I uploaded the updated DDF and restarted Deconz. In Home Assistant I still have the same issue, when I trigger the siren the switch automatically sets to off after a few moments, with the siren still making noise, and so I can't stop it from Home Assistant.

pawlizio, 2025-04-18:
Well, it has not solved the issue with the switch as I wrote in one of my previous posts, only that you can now at least use siren.turn_off / siren.turn_on action to control them. But also this is not very reliable. I have several smoke detectors and if I turn on the siren on all them, I need to fire several times the siren.turn_off action to ensure that all of them turn really off. However I assume that these are all different root causes.

By the way the switch / toggle also never worked for me in the past reliable. Switching them on and immediately off worked sporadically to turn the siren off again. I also have a siren entity for my ring doorbell. On ring integration the siren is not using a switch / toffle in the front end, but two seperate buttons. Probably this would be a solution for this as well.