dresden-elektronik/deconz-rest-plugin #8041: Sunricher HK-ZCC-ZLL-A: window covering device

kvanisterdael, 2024-11-25:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Zigbee Curtain Motor Controller SR-ZG9080A

### Manufacturer

Sunricher

### Model identifier

HK-ZCC-ZLL-A

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

kvanisterdael, 2024-11-25:
Device is exactly the same as the one defined in SR-ZG9080A.json.

kvanisterdael, 2024-11-25:
Proposal for DDF file: 
[hk-zcc-zll-a.json](https://github.com/user-attachments/files/17904563/hk-zcc-zll-a.json)

Smanar, 2024-11-29:
Hello, but the support is not native for this device ?

kvanisterdael, 2024-12-02:
No, in v 2.28.1 it didn't work out of the box like the previous device (SR-ZG9080A.json) did.

closed 2025-01-01 by github-actions[bot]