dresden-elektronik/deconz-rest-plugin #8042: Moes thermostat TS0601 - (_TZE200_ye5jkfsb) showing incorrect room temperature

welo78, 2024-11-25:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hello, it seems that a Moes electric heating thermostat has two issues.

1. Temperature is incorrectly displayed as 0 Celsius (or wrongly parsed/calculated) or 2 degrees when temperature is 22 in the room
2. Home assistant picks it up as increments of 0,5 degree when it can only make 1 degree steps

Thermostat in question: https://www.aliexpress.com/item/1005005963529004.html?spm=a2g0o.order_list.order_list_main.11.406c1802Nirlz9

### Steps to reproduce the behavior

1. Log into Conbee gateway
2. Check thermostat tab
3. Thermostat showing incorrect temperature

### Expected behavior

1. Display correct room temperature
2. Make increments of 1 degree step instead of 0,5 in home assistant (can be fixed in HA)

### Screenshots

[image]

### Environment

- Host system: Raspberry Pi 4
- Running method: Raspbian + Home Assistent deCONZ Add-on
- Firmware version:26400500
- deCONZ version: 2.28.1
- Device: RaspBee I 
- Is there any other USB or serial devices connected to the host system? If so: Z-Wave dongle

### Additional context

I am all up for testing any DDF if this is a unsupported device at the moment. However as it picked it up I kinda thought it is supported.

Smanar, 2024-11-26:
Perhaps https://github.com/Koenkk/zigbee2mqtt/issues/11980 ?

But it seem working for somes users, but not all.

We can try editing the DDF devices\moes\Moes_TZE200_ztvwu4nk_thermostat.json
How many TRV have you  ? because editing direclty this file will impact ALL yours devices using the same DDF

And can you check the value returned by your device ?

It's visible on logs in Deconz/help/Debug view, I think the flag "DDF" will be enought to see the value used by the device.

welo78, 2024-11-27:
Hello @Smanar thank you for the reply. I added the debug message to the original post.

I have 1 of this device for testing purposes. I am looking for swapping non-smart thermostat for zigbee operated.

As for DDF when I enabled debug view and ticked DDF it started dumping all kinds of DDFs it knows in a long list. Will this suffice?
[image]

closed 2024-12-27 by github-actions[bot]