dresden-elektronik/deconz-rest-plugin #8045: Namron motion sensor

skysurfer2000, 2024-11-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Namron Zigbee Motion Sensor

### Manufacturer

Namron AS

### Model identifier

4512763

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2024-11-29:
Hello can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Namron AS",
  "modelid": "4512763",
  "vendor": "Namron",
  "product": "Motion sensor (4512763)",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr",
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/presence",
          "awake": true
        },
        {
          "name": "state/tampered",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

skysurfer2000, 2024-11-30:
Sorry for being a noob. I tried to import this directly in the Deconz DDF editor and it failed. I also tried to import it in the toolbox, but it just keeps "spinning", nothing happens. Any suggestions?

Smanar, 2024-12-01:
Do you want to make the PR or want I make it ?

skysurfer2000, 2024-12-01:
> Do you want to make the PR or want I make it ?

I don’t know how to make a PR, so please make it. Thanks.

Smanar, 2024-12-01:
Done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8049

closed 2024-12-25 by skysurfer2000