dresden-elektronik/deconz-rest-plugin #8048: Namron Thermostat 4512758 16A

kimchiii0411, 2024-11-30:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

NAMRON ZIGBEE TERMOSTAT 16A

### Manufacturer

Namron

### Model identifier

4512758

### Device type to add

Thermostat

### Node info

[image]

### Endpoints and clusters

Electrical Measurement
[image]
[image]
[image]
Simple metering
[image]
Thermostat
[image]

### Basic

[image]

### Further relevant clusters

This has detailed zigbee information: English on page 13 and on. https://www.elektroimportoren.no/docs/lib/4512758-Brukerveiledning-5.pdf

Smanar, 2024-12-01:
You need to press the "read" button before taking the capture, for deconz aask for value.
BTW you can try this DDF, it's a defaut one for this kind of device.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "NAMRON AS",
  "modelid": "4512758",
  "vendor": "Namron",
  "product": "Thermostat",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "default": 0
        },
        {
          "name": "config/mode"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
[... 220 lines cut ...]
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0201",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 30,
          "max": 360,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

kimchiii0411, 2024-12-01:
I did press read. Maybe I forgot on one of the clusters. Can you see one
where I didn't? I'll try that DDF.

Smanar, 2024-12-01:
- Thermostat > local temperature is strange.
- Electrical Measurement > all value are bad

But don't worry, will see later if the DDF fail.

kimchiii0411, 2024-12-01:
That is probably because it's in regulator mode and not using temperature.
Electrical I did forget. 
[image]

kimchiii0411, 2024-12-01:
The DDF did not work. The zigbee part is well documented in the link I attached the request. I can help if there is a template that shows what values goes where roughly.

Smanar, 2024-12-03:
Have checked again the DDf here https://jsonlint.com/
No problem, can you check the file on your OS ?

kimchiii0411, 2024-12-04:
Copied from my file and that passes the linter as well. I added the file and --force-recreated the container. Still the same error 
[image]

Smanar, 2024-12-05:
Open the file with a text editor.

kimchiii0411, 2025-01-02:
I'm not sure why that would help. I created the file with nano and pasted the raw copy.

Smanar, 2025-01-03:
Because the synthax is good, tested on external site, so if deconz is not able to load it it's from the file.

kimchiii0411, 2025-01-03:
I created the file with nano inside the container the first time to test. That should be sufficient. I can also open it with nano on the host. What else would you like for me to try?

Smanar, 2025-01-04:
>I can also open it with nano on the host

Exactly that to be sure there is nothing corrupted. if the DDF is bad it's just ignored, for deconz fail to open it, need a typo for exemple.
Can try to take a look in the logs, to see a better error message, but not sure deconz will make it
Can try too with a certified working DDF and change just the manufactureName/Model ID, but it's exaclty what I have done to make this one.

You are using the deconz docker image ?

kimchiii0411, 2025-01-05:
It didn't work. I took one of the ones you provide from the /home/deconz/.local/share/dresden-elektronik/deCONZ/ folder and changed modelid to 4512758. Still failed to open. 

I'm using the deconz docker image. image: "ghcr.io/deconz-community/deconz-docker:latest"

Smanar, 2025-01-05:
And you are making your change in the folder "/opt/deconz", it's the folder outside the docker.
Try the reverse, save the "auto generated DDF" (even it's a draft status") and try to edit it or copy the DDF contain inside.

kimchiii0411, 2025-01-18:
I've mapped a folder to a ddf folder I have on the host. Does it matter which folder I put DDF files on in the container?

kimchiii0411, 2025-01-18:
Oh. That didn't work either. I saved the auto generated DDF. Failed to open that as well.

kimchiii0411, 2025-01-18:
Looks like it's an issue with deconz. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7971

Smanar, 2025-01-19:
Deconz load all DDF at start, so can try to reload deconz after a DDF file edition.

kimchiii0411, 2025-01-21:
I have reloaded multiple times and recreated.

closed 2025-01-29 by manup