dresden-elektronik/deconz-rest-plugin #8059: Human Presence Detector - WY-M100-S

brientim, 2024-12-12:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Smart Human Presence Motion Sensor With Luminance

### Manufacturer

_TZE204_laokfqwu

### Model identifier

TSO601

### Device type to add

Sensor

### Node info

The WZ-M100 is a clone of the ZY-M100 with revised versioning. It has the same capabilities and functions has the ZY-M100. 

As the new devices has the potential to have a different firmware, I have raised this as a new request. The alternative could be  to the existing MmwRadar Human Presence Detector (TS0601) JSON.

Product name: Human Presence Detector - WY-M100-S
[WZ_M100 MmwRadar Human Presence Detector (TS0601)json.json](https://github.com/user-attachments/files/18103800/WZ_M100.MmwRadar.Human.Presence.Detector.TS0601.json.json)

Manufacturer: _TZE204_laokfqwu
Model identifier: TS0601
Device type : Sensor

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

brientim, 2025-01-02:
Bumping this to see if it is picked up.

zimko, 2025-01-25:
`
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_laokfqwu",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "MmwRadar Human Presence Detector (TS0601)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
        },
        {
[... 240 lines cut ...]
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}

`

Whith this DDF sensor work but not luminosity

Mimiix, 2025-01-25:
@Smanar can you help:)?

zimko, 2025-01-25:
Find solution for luminosity : 

This ddf need this script : lux_to_lightlevel.js

Solution inside your custom directory (/root/.local/share/dresden-elektronik/deCONZ/) create directory /generic/illuminance_cluster
Copy the original file 'lux_to_lightlevel.js' ( /usr/share/deCONZ/devices/generic/illuminance_cluster/lux_to_lightlevel.js) to the new illuminance_cluster created first.

Enjoy.

Smanar, 2025-01-25:
I m seing there is 5 Tuya presence detector ^^ I think we can group some of them ?
Your DDf is based o this one ? https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/ZY-M100_human_breathing_presence.json

So can just add your model to the list ?

Do you want to make the PR or want I make it ?

brientim, 2025-01-25:
@Smanar 

Correct the DDF was based on the ZY-M100…uI believe that they could be appended to the original ZY-M100 DDF as that would ensure consistency. 

Can you please raise the PR and consolidate as required.

Smanar, 2025-01-25:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8104

Have not modified the path for the luminosity because it's good for file in the official folder `generic/illuminance_cluster/lux_to_lightlevel.js` is present for file in the tuya folder

brientim, 2025-01-26:
@Smanar  Acknowledges and thank you.

Mimiix, 2025-01-26:
Please keep it open until the pr is merged :)

closed 2025-01-29 by manup