dresden-elektronik/deconz-rest-plugin #8062: Livarno Lux E27-  CCT Bulb

Zazou89, 2024-12-12:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Livarno Lux E27 9W CCT Bulb

### Manufacturer

LIDL Livarno Lux

### Model identifier

HG06492C

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]
[image]

### Level Control
[image]

### Color Control
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2024-12-12:
Hello, but the support is not native for this kind of device ?

Zazou89, 2024-12-12:
> Hello, but the support is not native for this kind of device ?

I assumed it wasn't 100% integrated because I get warning messages for these devices in Homenridge (legacy code warning / XY gamut). 

Similarly, I don't know if this has anything to do with it, but in the DeconZ GUI it doesn't say "DDF" on the top right for these devices.

Smanar, 2024-12-13:
Nope, you are right, this device don't have DDF, but most of bulbs are working natively.

closed 2025-01-11 by github-actions[bot]