dresden-elektronik/deconz-rest-plugin #8070: TZ3000_5f43h46b

hubertgruber, 2024-12-19:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Mini Smart Socket

### Manufacturer

_TZ3000_5f43h46b

### Model identifier

TS011F

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering
[image]

### Electrical Measurement
[image]

### Any other cluster of relevance/interest
[image]

hubertgruber, 2024-12-19:
It seems to be identical to this device:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5633#issuecomment-1008266360
But with a different manufacturing ID.

hubertgruber, 2024-12-21:
https://www.aliexpress.com/item/1005007116126515.html
[image]
It's the same Tuya plug as is mentioned in #5633.

Smanar, 2024-12-22:
Can try the same DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZ3000_gjnozsaz",
    "_TZ3000_w0qqde0g",
    "_TZ3000_zloso4jk",
    "_TZ3000_rdfh8cfs",
    "_TZ3000_fgwhjm9j",
    "_TZ3000_waho4jtj",
    "_TZ3000_xzhnra8x",
    "_TZ3000_5f43h46b"
  ],
  "modelid": [
    "TS011F",
    "TS011F",
    "TS011F",
    "TS011F",
    "TS011F",
    "TS011F",
    "TS011F",
    "TS011F"
  ],
  "vendor": "Neo",
  "product": "Smart plug 16A with power monitoring EU (NAS-WR01B)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "eval": "Item.val = 10*Attr.val"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 360,
          "default": 0
        },
        {
          "name": "state/voltage",
          "refresh.interval": 360,
          "default": 0
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 5,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 15,
          "max": 500,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 15,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 5,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

```

hubertgruber, 2024-12-23:
> Hello, can try the same DDF
> 
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": [
>     "_TZ3000_gjnozsaz",
>     "_TZ3000_w0qqde0g",
>     "_TZ3000_zloso4jk",
>     "_TZ3000_rdfh8cfs",
>     "_TZ3000_fgwhjm9j",
>     "_TZ3000_waho4jtj",
>     "_TZ3000_xzhnra8x",
>     "_TZ3000_5f43h46b"
>   ],
>   "modelid": [
>     "TS011F",
>     "TS011F",
>     "TS011F",
>     "TS011F",
>     "TS011F",
>     "TS011F",
>     "TS011F",
>     "TS011F"
>   ],
>   "vendor": "Neo",
>   "product": "Smart plug 16A with power monitoring EU (NAS-WR01B)",
>   "sleeper": false,
>   "status": "Gold",
>   "subdevices": [
>     {
>       "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
>       "restapi": "/lights",
>       "uuid": [
>         "$address.ext",
>         "0x01"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion",
>           "parse": {
>             "fn": "zcl:attr",
>             "ep": 1,
>             "cl": "0x0000",
>             "at": "0x0001",
>             "script": "../tuya/tuya_swversion.js"
>           },
>           "read": {
>             "fn": "zcl:attr",
>             "ep": 1,
>             "cl": "0x0000",
>             "at": "0x0001"
[... 204 lines cut ...]
>           "min": 15,
>           "max": 500,
>           "change": "0x00000001"
>         },
>         {
>           "at": "0x0508",
>           "dt": "0x21",
>           "min": 15,
>           "max": 300,
>           "change": "0x00000001"
>         },
>         {
>           "at": "0x050B",
>           "dt": "0x29",
>           "min": 5,
>           "max": 300,
>           "change": "0x00000001"
>         }
>       ]
>     }
>   ]
> }
> ```

Thanks! I will research how to do this in the Home Assistant plug-in and then make a PR if it works.

Smanar, 2024-12-23:
Not easy on HA but to make short need to use the hidden folder `/data/.local/share/dresden-elektronik/deCONZ/devices/`

https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

hubertgruber, 2024-12-30:
If I use this service:
https://deconz-community.github.io/ddf-tools/#/auto-bundler
Which files do I need to upload there to create a bundle? The add-ons front end seems to accept these.

sanluca, 2025-01-05:
I have the same smart plug but with a different vendor ID: _TZ3000_ww6drja5

Can it be added in the next update?

Smanar, 2025-01-05:
Hello, can you test the DDF to see if it works ?

sanluca, 2025-01-05:
> Hello, can you test the DDF to see if it works ?

Hi, 
Yes, it's working for me. I have two of these plugs, I simply added them to the device list in the JSON file above

Smanar, 2025-01-05:
>I simply added them to the device list in the JSON file above

What are your model id ? (to add them too)
Do you want to make the PR to add them or want I make it ?

sanluca, 2025-01-06:
If you can, please add them yourself, it's faster, thanks

_TZ3000_ww6drja5

TS011F

Smanar, 2025-01-06:
Added https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8081

closed 2025-02-04 by github-actions[bot]