dresden-elektronik/deconz-rest-plugin #8073: Heimgard Smart Lock

donnib, 2024-12-21:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Heimgard Smart Lock

### Manufacturer

Heimgard Technologies

### Model identifier

HT-SLM-2

### Device type to add

Lock

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Smanar, 2024-12-21:
Hello can try this DDF 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Heimgard Technologies",
  "modelid": "HT-SLM-2",
  "vendor": "Heimgard",
  "product": "Door lock (HC-SLM-2)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DOOR_LOCK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0101"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val/2",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val ==1",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'not fully locked' } else if (Attr.val == 1) { Item.val = 'locked' } else if (Attr.val == 2) { Item.val = 'unlocked' } else { Item.val = 'undefined' }",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val !=1",
            "fn": "zcl:attr"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 5,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 5,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Have mimic an actual one, not sure for battery as on your capture both lines are at "0".

donnib, 2024-12-21:
Thank you, i can't load the file. I tried following :

1. Make a new file on the rpi with the contents above and save it as json
2. Open the node in deconz ui, edit DDF
3. Click File -> Open and choose the json file from step 1
-> Failed to open 

What am i doing wrong ?

donnib, 2024-12-22:
[image]

Smanar, 2024-12-22:
What is you  OS ?
The json need to be in a "devices" folder, then just restart deconz, never use the editor, except to test/check.

> 
All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired

donnib, 2024-12-22:
I am running deconz on a rPi and i believe its a Debian distribution. I wanted to test the file first before restarting second. Why is that not possible ?

Smanar, 2024-12-22:
Deconz load DDF only at start and too much problem using the editor.

donnib, 2024-12-25:
I have a folder called : /.local/share/data/dresden-elektronik/deCONZ/devices
should i call the file a specific name because i tried making a file called lock.json and place the content above and restarted deconz but nothing happened to the lock, do i need to repair the lock again or ?

Smanar, 2024-12-25:
No, it just need to be in the folder "devices", can make a subfolder caller Heimgard.
And yes you need to pait it again, and this time deconz will use the DDF to reconise it, if you edit the DDF later it will be not necessary, but it is for the first inclusion (ATM the device is not reconised, so not in the API at all)

donnib, 2024-12-25:
Got it, i repaired and now it shows up however i was hoping i would get events from the lock when it's unlocked like which code was used to unlock to identify the person but i don't see any events in HA either because the lock doesn't send them or something else is needed more in the DFF ? (i don't know if DFF files have any implications on this)

Smanar, 2024-12-25:
Nope the RFID (or code gestion) need this kind of PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5918
The working mode ATM is realy basic.

closed 2025-01-23 by github-actions[bot]