dresden-elektronik/deconz-rest-plugin #8075: Tuya Smart Scene Switch 4-Gang Battery

YvesSun23, 2024-12-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Smart Scene Switch 4-Gang

### Manufacturer

_TZ3000_0ht8dnxj

### Model identifier

TS004F

### Device type to add

Switch

### Node info

I try to include a Tuya compatible 4 gang scene switch (12 scenes).
4 buttons with single press, double press and long press.
I found on my RPi with ConbeeII the _TZ3000_TS004F_switch.json but it's only a single switch.

[image]

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

[image]

### On/Off

[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2024-12-27:
Hello what is your OS ?
You need to add your model to this DDF for exemple https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_ja5osu5g_smart_button.json
But need to edit the buttonmap file, and not easy on all OS.

YvesSun23, 2024-12-27:
Hi, it's Linux running on a raspberry.
Where can I find the button map file?

Smanar, 2024-12-27:
Ha nice so it's easy, on my side I have it in `/usr/share/deCONZ/devices/button_maps.json`
Just add your model to this list
```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_yw5tvzsk", "_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu","_TZ3000_wqcbzbae", "_TZ3000_pkeqinnt", "_TZ3000_uaa99arv", "_TZ3000_wbfgbpxq", "_TZ3000_fa9mlvja", "_TZ3000_tk3s5tyg", "_TZ3000_mrpevh8p", "_TZ3000_qgwcxxws", "_TZ3000_yj6k7vfo", "_TZ3000_axpdxqgu", "_TZ3000_a4xycprs", "_TZ3000_u3nv1jwk"],
            "map": [
```
And restart deconz after having edited the DDF too.

YvesSun23, 2024-12-27:
Thank you for your assistance. The system is generally working, but I am experiencing some issues that I may not fully understand.

1. In the Phoscon app, the mapping for on/off functions for all four keys is functioning correctly. However, I would like to know if it is possible to implement a feature similar to that of Ikea switches, where a single press turns the light on and a long press dims it up, among other functionalities.

[image]

[image]

2. Additionally, in HomeKit, it only shows as a single switch with options for single press, double press, and long press.

[image]

[image]

YvesSun23, 2024-12-27:
I have used the _TZ3000_ja5osu5g_smart_button.json file

Smanar, 2024-12-28:
I m sorry but I don't know phoscon, and IDK how this device is handled by their code.
If the code is working you need to have diffeent values on state/buttonevent field, with value like X00Y, X is the button number and Y the action.

YvesSun23, 2024-12-28:
ok, that's no problem I will try it by myself.
Thank you very much for your help :)

closed 2025-01-27 by github-actions[bot]