dresden-elektronik/deconz-rest-plugin #8082: Ledvance SMART+ PLUG COMPACT EU EM T

thba17, 2025-01-08:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

LEDVANCE Zigbee SMART+ Compact outdoor Plug EU - Energy meter

### Manufacturer

LEDVANCE

### Model identifier

PLUG COMPACT EU EM T

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

### Simple Metering
[image]
[image]

### Electrical Measurement
[image]
[image]
[image]
[image]
[image]

### Any other cluster of relevance/interest
### Diagnostics
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### OTAU
[image]

thba17, 2025-01-20:
Maybe "Device type to add" should be Plug as it is a smart plug. On/Off can be controlled from HA.
I see there is a related #7923 whereas this new one has energy metering capabilities.

fredl99, 2025-02-12:
> Maybe "Device type to add" should be Plug

Deconz lists these plugs under "lights", so I guess that's okay. To my understanding (compared with json for other devices like from Bosch) there has to be a subdevice added, which should then show up under 'sensors'.

fredl99, 2025-02-14:
There's something new: #8123

thba17, 2025-03-10:
As mentioned before, from my side this topic is not stale.

thba17, 2025-04-01:
Still not stale from my pov.
Is anyone checking in on this?
Will it be handled or dismissed?
Total silence.
Some form of feedback would be appreciated.

manup, 2025-05-02:
Closing for now as completed in v2.29.3-beta via https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8123.

closed 2025-05-02 by manup