dresden-elektronik/deconz-rest-plugin #8085: _TZ3000_0hkmcrza - TS0203 - Tuya Zigbee Door Window Sensor ZD06

Th30n34nd0nl7, 2025-01-11:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Zigbee Door Window Sensor ZD06

### Manufacturer

_TZ3000_0hkmcrza

### Model identifier

TS0203

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### Identify
[image]

### Power Configuration
[image]

### IAS Zone
[image]
[image]

### Further
Hi, just got some new door/window sensors, that won't appear in Phoscon. It seems they are similar to other TS0203 devices and that just the manufacturer is different although I am new to the whole DDF world. 

@Smanar seems to be the right person to contact, so I linked you here.
Looking forward to diving deeper into DDFs 😎

Th30n34nd0nl7, 2025-01-11:
I got it working by editing the following file:
`/usr/share/deCONZ/devices/tuya/_TZ3000_TS0203_door_sensor.json`

Here I just added "_TZ3000_0hkmcrza" into the "manufacturername" string array and "TS0203" into "modelid" string array.

Seems to be just another manufacturer clone.

closed 2025-01-11 by Th30n34nd0nl7

Smanar, 2025-01-12:
Nice, do you want to make the PR to add it officialy or want I make it ?

Th30n34nd0nl7, 2025-01-14:
Hi, I've never done a PR before, so please feel free...

Smanar, 2025-01-15:
Done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8092
Thx