dresden-elektronik/deconz-rest-plugin #8086: Moes Smart Dimmer Module 1 Gang   _TZE204_hlx9tnzb

c4coer, 2025-01-11:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Smart Dimmer Swtich

### Manufacturer

Moes 

### Model identifier

ZS-SR-EUD

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

c4coer, 2025-01-11:
I tried this ddf, but the power variation does not work, just on and off. thanks in advance for the help
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7703

mangoschorle, 2025-01-15:
@c4coer I can reproduce this issue! On/Off works , power variation does not work

c4coer, 2025-01-15:
you epux tried this dff, we can see with the variation when we press the buttons directly, but the control via the application does not work in the end just works the on/off

c4coer, 2025-01-15:
{
  "schema": "devcap1.schema.json",
  "uuid": "3c38d64a-5192-4183-882c-e4320e914bb5",
  "manufacturername": "_TZE204_hlx9tnzb",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Smart Dimmer Module (TS0601)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/tuya_unlock"
        },
		{
          "name": "state/alert",
          "default": "none",
          "public": false
        },
        {
          "name": "state/on",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 1,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;"
          },
          "read": {
            "fn": "tuya"
          },
          "refresh.interval": 300
        },
        {
          "name": "state/bri",
          "parse": {
            "fn": "tuya",
            "dpid": 2,
            "eval": "Item.val = (Attr.val / 1000.0) * 254.0;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 2,
            "dt": "0x2b",
            "eval": "(Item.val / 254.0) * 1000.0;"
          },
          "read": {
            "fn": "none"
          },
          "refresh.interval": 42000
        },
        {
          "name": "state/reachable"
		}
      ]
    }
  ]
}

c4coer, 2025-01-15:
@mangoschorle mangoschorle
 you can try the ddf above and tell me if it works for you

c4coer, 2025-01-15:
I think the best thing is to do both.

closed 2025-02-13 by github-actions[bot]