dresden-elektronik/deconz-rest-plugin #8087: Webfrontend not reachable in versions 2.27.6 and 2.28.1

FrankH1, 2025-01-12:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hi there,
I'm running a Conbee II Stick (FW 26780700) on a Raspi4 Model B with Openhab 4.
Today updated manually the deconz software - was running 2.24.3 and was going to upgrade to the latest version.
Although the update worked and the stick delivers data to Openhab - I can't access it anymore via a browser. (Checked and changed of course the port to 8081). Using https://phoscon.de/pwa delivers the Phoscon-GW with its IP adress - but I can't get in using http://IP:8081/.
Downgraded step by step and it was again working with 2.26.3.
Is there anything I'm doing wrong?
Cheers Frank

### Steps to reproduce the behavior

manual upgrade as described here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

### Expected behavior

Access to webinterface/Gateway on http://IP:8081/.

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian 
- Firmware version: 26780700)
- deCONZ version : 2.27onwards
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: external card reader?

Smanar, 2025-01-12:
Hello

>Using https://phoscon.de/pwa delivers the Phoscon-GW with its IP adress - but I can't get in using http://ip:8081/

It mean it works using `https://phoscon.de/pwa` ?

FrankH1, 2025-01-12:
No - it comes up with the symbol of the gateway with its IP adress under the picture - but thats it.
[image]

FrankH1, 2025-01-12:
Meanwhile I found a similar issue with earlier versions - 
....[https://forum.phoscon.de/t/phoscon-de-pwa-login-html-does-not-work-anymore/3967/92](url)
but I dont find a solution there for me

MSchmitt67, 2025-01-13:
Hi,
try this
http://ip/pwa/login2.html?

FrankH1, 2025-01-13:
I'm completely puzzled now. Did upgrade again to deconz-latest in order check the question from MSchmitt67.
Its working and I get access via the IP:8081 adress, as screenshot shows
[image]

What I don't get is the login page using the pwa link
So if I try 
[image]
but I cannot login.
If I use the IP adress with port (http://192.168.200.199:8081/pwa/login2.html?)  I immediately land on a nearly similar page (with "undefined" in the bottom left and the correct Gateway picture of the Conbee stick) - but there I can click on the Gateway and I'm in (even without typing the pwd).
[image]
So at least its working now for me - but still there is the fear to loose access if the ~pwa stuff is not working

MSchmitt67, 2025-01-14:
Hi,

have you tried without the port ?

http://192.168.200.199/pwa/login2.html?

FrankH1, 2025-01-14:
website not reachable
[image]

closed 2025-02-12 by github-actions[bot]