dresden-elektronik/deconz-rest-plugin #8089: Device Request - Eurotronics Zigbee Comet

altussulta, 2025-01-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Eurotronics Zigbee Comet

### Manufacturer

Eurotronics

### Model identifier

SPZB0001

### Device type to add

Thermostat

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2025-01-13:
Device is somewhat supported, but determined to have faulty firmware. Please check the existing issues.

altussulta, 2025-01-14:
My bad, I didn't search for the model identifier, but although the identifier is the same, this is the Comet and not Spirit, perhaps the firmware issue has been resolved?
I am able to control the target temperature, but I am trying to achieve direct valve control or use a remote temperature sensor. Right now the temperature reports about 2 degrees low when off and 2 degrees high when on.

Is this something that can be achieved here? When I try to write "set valve position" with TRV mode manual and system mode on heat (which should be manual control) it just reverts back to 0.

Any help would be greatly appreciated!

Mimiix, 2025-01-14:
If it has the same model id / manufacturer name, we can't know the difference. As far as i can see now, this is the exact same device (other than the software version date). I'll ask the devs to check in.

altussulta, 2025-01-16:
I understand, and I hope something can be done. Thanks for checking with the developers.

closed 2025-02-15 by github-actions[bot]