dresden-elektronik/deconz-rest-plugin #8098: macOS Presence Sensors always report "no motion"  / false on event stream when the sensor actually detects motion

doodahdayz, 2025-01-20:
### Does the issue really belong here?

- [x] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

When a motion sensor detects motion the event reported via the web socket will report that the motion detected is false.  This only occurs with the November macOS build when trying to run the Conbee III on macOS.  It reports correctly when running Conbee on a Linux box.  

### Steps to reproduce the behavior

Run macOS version of deConz.
Have a motion sensor enabled and running on the system.
Cause the motion sensor to detect motion and watch the event stream on the web socket. It will report that the sensor fired off an event, but it will report false for motion detected / presence .... 

### Expected behavior

When the sensor detects motion the web socket even should indicate that in fact there was a presence detected. 

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / deconz-docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

Mimiix, 2025-01-20:
Please add the environment information

doodahdayz, 2025-01-21:
This is running on M4 Mac Mini that is running macOS Sequoia 15.2
Since this is Apple Silicon machine it is also being run via Rosetta.  Since Apple no longer sells any Intel based machines it would be nice if deConz could be compiled for Apple Silicon.

Mimiix, 2025-01-21:
> This is running on M4 Mac Mini that is running macOS Sequoia 15.2 Since this is Apple Silicon machine it is also being run via Rosetta. Since Apple no longer sells any Intel based machines it would be nice if deConz could be compiled for Apple Silicon.

Please update your initial post with the correct info

doodahdayz, 2025-01-21:
deConz Gateway Version is 2.29
Runing a Conbee III which is attached to with an extension cable
Firmware is 26530900

I don't know what else I can tell you.    I cannot edit the "Environment" section... but I have now listed everything I can about the "environment".   

It boils down to the motion sensors do not correctly report when the sensor detects motion on this platform.  So it has forced me back to using a Raspberry Pi until this issue can be fixed and hopefully compiled for Apple Silicon.

Mimiix, 2025-01-21:
https://docs.github.com/en/issues/tracking-your-work-with-issues/using-issues/editing-an-issue

Will help you :)

doodahdayz, 2025-01-21:
All the info has been provided.  Obviously the macOS version of deConz is a red headed step child.  I don’t have time to keep updating this with the info that is already there.  If no one wants to be bothered to fix it, then so be it.   I opened the issue for others.. I’ve already moved back to Linux as I don’t want to wait around for someone to even care about macOS.

Regards,
Dan

closed 2025-02-19 by github-actions[bot]