dresden-elektronik/deconz-rest-plugin #8101: ZigBee Smart Thermostat Programmable Temperature Controller for Electric Floor Heating

olskar, 2025-01-22:
### Is there already an existing issue for this?

- [x] I have searched the existing issues and there is none for my device

### Product name

ZigBee Smart Thermostat Programmable Temperature Controller for Electric Floor Heating

### Manufacturer

tuya

### Model identifier

TS0601

### Device type to add

Thermostat

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2025-01-23:
Hello, the previous version is here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/moes/Moes_TZE200_ztvwu4nk_thermostat.json
Can try with just adding your model to the list to see if some part are common or not.

olskar, 2025-01-24:
Can I do this change myself?

Smanar, 2025-01-24:
Sure, depend of your OS, if you have a classic OS, you can edit direclty the file

`All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired`

One is persistent, the other will be removed at next deconz update.

If you have HA or docker, you need to use the persistent folder and create a new file.
In HA >` /data/.local/share/dresden-elektronik/deCONZ/devices/` https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/19
In docker > `opt/deCONZ/devices/`

olskar, 2025-01-25:
Thanks, added the model to the list. On and off works fine, but nothing else, no temperature reported and no way to change the temperature :(

Smanar, 2025-01-25:
You are right, sorry lot of dpid have moved.
Can you rollback your last changes and create this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_xalsoe3m",
  "modelid": "TS0601",
  "vendor": "Moes",
  "product": "ZigBee Smart Thermostat Programmable Temperature Controller for Electric Floor Heating",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {
            "off": 0,
            "heat": 1
          }
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "../tuya/tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {
            "fn": "tuya",
            "dpid": 50,
            "eval": "Item.val = Attr.val * 100;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 50,
            "dt": "0x2b",
            "eval": "Item.val / 100;"
          },
          "read": {
            "fn": "tuya"
          },
          "refresh.interval": 84000
        },
        {
          "name": "config/locked",
          "parse": {
            "fn": "tuya",
            "dpid": 39,
            "eval": "Item.val = Attr.val;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 39,
            "dt": "0x10",
            "eval": "Item.val;"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
            [
              "off",
              0
            ],
            [
              "heat",
              1
            ]
          ],
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'off' } else { Item.val = 'heat' }"
          },
          "write": {
            "fn": "tuya",
            "dpid": 1,
            "dt": "0x10",
            "eval": "if (Item.val == 'off') { false } else { true }"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "config/preset",
          "parse": {
            "fn": "tuya",
            "dpid": 2,
            "script": "../tuya/tuya_trv_preset.js"
          },
          "write": {
            "fn": "tuya",
            "dpid": 2,
            "dt": "0x30",
            "script": "../tuya/tuya_trv_preset_set.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {
            "fn": "tuya",
            "dpid": 7,
            "eval": "Item.val = Attr.val * 10;"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "state/valve",
          "parse": {
            "fn": "tuya",
            "dpid": 47,
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          }
        }
      ]
    }
  ]
}
```

olskar, 2025-01-25:
Thank you! 
Progress :) 

Now I can select desired temperature, and turn it on and off, but it still does report current temp as 0 degreess

Smanar, 2025-01-25:
```
        {
          "name": "state/temperature",
          "parse": {
            "fn": "tuya",
            "dpid": 16,
            "eval": "Item.val = Attr.val * 10;"
          },
          "read": {
            "fn": "none"
          }
        },
```

and with DPID = 16 ?

olskar, 2025-01-27:
That did work :)

closed 2025-02-22 by olskar